PX4/PX4-Autopilot #6027: multicopter: uncommanded yaw in stabilize mode

kd0aij, 2016-12-11:
observed on both v1.5.1rc4 and recent master
[image]
http://logs.px4.io/plot_app?log=d3b324da-5504-48ff-b613-e05be010063e

[image]
http://logs.px4.io/plot_app?log=21664518-008c-48f7-9340-1e3adf34e4aa

kd0aij, 2016-12-11:
This behavior results from using the internal magnetometer, which develops a 90 degree error in heading at hover thrust levels. With LPE, the effect is immediate: the copter yaws (smoothly) 90 degrees to the left on takeoff. With EKF2, there is a delay (15 to 30 seconds in the above plots), then a much more abrupt yaw reaction, which sometimes exceeds 90 degrees. 

In this Spedix S250AQ, the Pixracer is hardmounted to the power distribution board (using the standard mounting holes) and 6mm standoffs. Longer standoffs could be used, but there would still be a relatively large thrust related error in the mag reading. It may be preferable to just disable the magnetometer for applications not using GPS and external mag.

LorenzMeier, 2016-12-11:
@kd0aij Here is a short writeup how you probably should solve this for all angles, not just yaw: https://github.com/PX4/Firmware/issues/6029

LorenzMeier, 2016-12-11:
I'm closing this one as #6029 tracks this already.

closed 2016-12-11 by LorenzMeier