PX4/PX4-Autopilot #6028: Ensure OS and Flight app versions are reported correctly

LorenzMeier, 2016-12-11:
@bkueng We need to ensure that the OS and flight app version is reported correctly in these locations:

  * ```ver all``` on the console
  * via MAVLink
  * via `logger` on http://review.px4.io

The goal is to have a table in our flight reporting tool which allows us to make statements like "NuttX 7.4 has 100 flight hours across all airframes". The related issue for the table is: https://github.com/PX4/flight_review/issues/8

We might need a second table with the flight hours accumulation. Described here: https://github.com/PX4/flight_review/issues/9

dagar, 2017-04-11:
Done?

bkueng, 2017-04-12:
Yes, except for the NuttX version. @davids5 ?

LorenzMeier, 2017-04-30:
Fixed

closed 2017-04-30 by LorenzMeier