PX4/PX4-Autopilot #6029: User-feedback for gyro drift compensation for racers

LorenzMeier, 2016-12-11:
@anton-matosov @kd0aij @priseborough The best way to estimate gyro drift is to use the human feedback model: On a racer the long-term average rate for roll and pitch is zero (because even to fly a circular course the roll rate is on average zero, the roll angle is not though). You can probably also make this statement for yaw in body frame.

I did design 2010 an attitude estimator using this knowledge (I didn't come up with this myself, my advisor back then shared the insight) which allowed attitude control with human in the loop even though I had no accel.

It might be quite worthwhile applying this as model for racers, as it is far more reliable than accels or mags. Its essentially "vision based" through the pilot.

kd0aij, 2016-12-11:
Interesting idea, but my experience with MatrixPilot says that just eliminating the magnetometer results in acceptable yaw behavior, while recent testing of the Q attitude estimator shows that it is accurate for roll/pitch even during sustained coordinated turns.

priseborough, 2016-12-11:
This was a manoeuvring flight without GPS which is something ekf2 has had minimal testing for as it has been designed as an inertial navigation system. My thinking is that for applications where only an attitude estimate is required and GPS and mag data are not available, ekf2 is not the estimator we should be using.

kd0aij, 2016-12-11:
Agreed, I was just getting ready to do a test flight using the Q estimator with mag disabled, to make sure we haven't had any regressions.

LorenzMeier, 2016-12-11:
@kd0aij Can you propose a PR to enable LPE for racers? And disable the fallback to internal mag.

kd0aij, 2016-12-11:
OK, will work on validating a no GPS/mag LPE config with S250 and QAV250.

kd0aij, 2016-12-11:
Flight log with magnetometer ignored in Q estimator: http://logs.px4.io/plot_app?log=14509588-ca34-482d-92f1-17bb53780b1a

Total yaw drift of about 50 degrees in 5 minutes; would not be noticeable when flying circuits in stabilized or acro mode. 

Plot below shows pilot yaw inputs required to hold east heading, and throttle inputs required to hold altitude within one or two meters while in ALTCTL. Note that baro temp drops about 11C during the flight; this probably affects the gyro offsets.

[image]

anton-matosov, 2017-01-17:
I have finally captured uncontrolled forward tilt on my Pixracer
http://logs.px4.io/plot_app?log=b81a4f42-d4a4-4270-8945-57fe07e93eb2

kd0aij, 2017-01-17:
@anton-matosov Is that log from a 250 and is the pixracer hard-mounted? Your rate P gains are quite low, as is the vibration level. Also, did attitude control performance suddenly go from good to bad?

anton-matosov, 2017-01-19:
Pixracer is semi-hard mounted. If I add P gain I get heavy oscillations. Controls are very precise, the only thing is that drone keeps rotating forward (slowly but surely, with very constant speed)
The issue is not consistent. A simple power cycling helps. I get this strange behavior only 1 out of 5-10 boots. The biggest concern is that you have to start flying before you know that there is an issue.
And I didn't have it for several month until I had to recalibrate my pixracer after the fix of https://github.com/PX4/Firmware/issues/6067
I had it back in July/Aug, but then it suddenly disappeared and reappeared recently.

kd0aij, 2017-01-19:
@anton-matosov I'd really like to understand why your P gains are so much lower than my S250AQ gains. I have 2300KV motors with 5x4.6 2-blade props (also 3S power) and am flying at an altitude of about 1800 meters.  FC is a hard-mounted Pixracer.

My PID parameters are in the header of this plot: https://github.com/kd0aij/impulse-response/blob/master/sample_ulogs/1KHz/S250/flight3/impulse_response.png and rate P gain is 0.19

 You can see my vibration levels here: https://github.com/kd0aij/impulse-response/blob/master/sample_ulogs/1KHz/S250/flight3/accel_sig.png

The full log is here: http://logs.px4.io/plot_app?log=9483a288-b429-43c3-bc17-f4828216ca51 but you'll have to use flightplot to display the sensor_accel and sensor_gyro records (flight review only looks as sensor_combined, and that's not logged here in order to reduce bandwidth).

anton-matosov, 2017-01-20:
@kd0aij I have no idea why do I have such a low P gains. With 3 blade 5051 props I have to lower my P even more.
I am going to try and swap motors on my 250 and see how it affects P gains.

LorenzMeier, 2017-01-20:
We merged a change on master which reduces the rate delay. Test appreciated.

LorenzMeier, 2017-07-18:
This is stale. Performance improved recently considerably, discussion moved to here: https://github.com/PX4/Firmware/issues/6780

closed 2017-07-18 by LorenzMeier