PX4/PX4-Autopilot #6033: Gyro Drift

tops4u, 2016-12-11:
Since 1.5.x I have rather often 

> PreFlight Check: Gyro Sensors inconsitent - check 

This has not happend before and when I look at the Telemetry Gyro Values I get: 

[image]

When I do Gyro Calib all the three Gyro values will be over each other... until next restart, where I get the warning again and the values being apart as shown above.

LorenzMeier, 2016-12-11:
We've relaxed those checks in 1.5.2 - please flash 1.5.2 / latest stable again. We will ultimately fix this using temperature compensation.

tops4u, 2016-12-11:
I'm already running 1.5.2.

dagar, 2016-12-11:
@tops4u the params were relaxed a bit. You might still have the old values.
Check COM_ARM_EKF_GB

https://github.com/PX4/Firmware/commit/1fbc688757971cc7ac371949986cd11032b80d41

tops4u, 2016-12-12:
@dagar what should I expect there when running 1.5.2?

dagar, 2016-12-12:
COM_ARM_EKF_GB was changed from 3.5e-4 to 8.7e-4.

tops4u, 2016-12-12:
I have some difficulties telling you the actual value I have ... 

[image]

dagar, 2016-12-13:
If you're familiar with the console you can see them there `param show COM_ARM_EKF*`, otherwise try reset to default and see if that silences the warning.

tops4u, 2016-12-13:
Thanks, I'll try to get the Value on the Console. Note that for QGC Users there is no Default Value set, and that most probably such small Values can not be set anyway.

mhkabir, 2016-12-13:
@tops4u You need to flash via QGC for it to load the parameter metadata properly. The defaults are there all right.

I would suggest flashing afresh and doing a full parameter reset and reconfiguring everything.

tops4u, 2016-12-13:
I'm confused. I'm always flashing via QGC as this is my only GroundStation Software I use. Currently 3.0.2. I'll try to flash again and have a look on the console for this Param as well. Probably today evening.

tops4u, 2016-12-16:
So finally I could do this. Precision on the command Line is not much better either:

`Symbols: x = used, + = saved, * = unsaved
x   COM_ARM_EKF_POS [40,46] : 0.5000
x   COM_ARM_EKF_VEL [41,47] : 0.5000
x   COM_ARM_EKF_HGT [42,48] : 1.0000
x   COM_ARM_EKF_YAW [43,49] : 0.5000
x   COM_ARM_EKF_AB [44,50] : 0.0050
x   COM_ARM_EKF_GB [45,51] : 0.0009`

So it should already have the new Value.

tops4u, 2016-12-17:
I still have this Message at every boot (also with current Master). Does this Printout from the Sensors reveal something unusual?

> INFO  [lib__ecl] validator: best: 0, prev best: 0, failsafe: NO (0 events)
> INFO  [lib__ecl] sensor #0, prio: 255, state: OK
> INFO  [lib__ecl] .val:  -0.0049, lp:  -0.0030 mean dev:   0.0000 RMS:   0.0010 conf:   1.0000
> INFO  [lib__ecl] .val:  -0.0026, lp:  -0.0130 mean dev:  -0.0000 RMS:   0.0022 conf:   1.0000
> INFO  [lib__ecl] .val:   0.0139, lp:   0.0013 mean dev:  -0.0000 RMS:   0.0014 conf:   1.0000
> INFO  [lib__ecl] sensor #1, prio: 75, state: OK
> INFO  [lib__ecl] .val:  -0.0114, lp:  -0.0060 mean dev:  -0.0000 RMS:   0.0021 conf:   1.0000
> INFO  [lib__ecl] .val:   0.0080, lp:  -0.0005 mean dev:  -0.0000 RMS:   0.0044 conf:   1.0000
> INFO  [lib__ecl] .val:  -0.0796, lp:  -0.0764 mean dev:  -0.0000 RMS:   0.0058 conf:   1.0000

tops4u, 2016-12-26:
This Problem is still persisting. Today I was denied Arming because of a to large Gyro drift. Apparently there really is a change in the Sensor Params, when I do calibrate. 

Left before, right after calibration:
```

CAL_GYRO0_XOFF	0.013286806	CAL_GYRO0_XOFF	0.016934713
CAL_GYRO0_YOFF	0.023283556	CAL_GYRO0_YOFF	0.041415602
CAL_GYRO0_ZOFF	0.013159839	CAL_GYRO0_ZOFF	0.011229334
CAL_GYRO1_XOFF	0.0139098	CAL_GYRO1_XOFF	0.02227083
CAL_GYRO1_YOFF	0.02918716	CAL_GYRO1_YOFF	0.031834483
CAL_GYRO1_ZOFF	-0.041347295	CAL_GYRO1_ZOFF	0.07019172
```

LorenzMeier, 2017-03-13:
I would recommend to do a temperature calibration run with the newly merged temp cal: https://dev.px4.io/sensor-thermal-calibration.html

closed 2017-03-13 by LorenzMeier