PX4/PX4-Autopilot #6037: Intel Aero CPU load & timings

LorenzMeier, 2016-12-12:
@lucasdemarchi @lvale The normal timing for PX4 is looking like this:

[image]

From this log:
http://review.px4.io/plot_app?log=1253eec0-17f1-4051-b858-dce51b268928

Aero looks quite different:
http://logs.px4.io/plot_app?log=a24d89aa-d930-4894-9c6e-65ae449476ba

We need to work out what is inducing the load and churn on Aero. Its not the logging subsystem but must be something else.

LorenzMeier, 2016-12-12:
@bkueng Any creative ideas?

LorenzMeier, 2016-12-12:
@davids5 Any ideas from your side?

bkueng, 2016-12-12:
One explanation for the generally high CPU is log streaming, in particular sending of the UART bytes. I asked @davids5, and we need UART TX DMA for this.
I'm not quite sure about the increases, but it could simply be that logging rate increases on arming, meaning more bytes need to be sent.

lucasdemarchi, 2016-12-12:
@bkueng we need logger disabled by default and the mavlink rate back to the value it was before. So basically I think it's about reverting

LorenzMeier, 2016-12-12:
@lucasdemarchi Could you do that, connect the mavlink shell and run top? If it drops then back to 60% load we should be fine. If not its something else.

davids5, 2016-12-13:
@LorenzMeier - I think I would need HW to dig into this.

LorenzMeier, 2016-12-14:
@lucasdemarchi Did you end up doing that experiment? `top` via MAVLink shell would tell you about the load.

lucasdemarchi, 2016-12-14:
@LorenzMeier I can do it... when I have the hw again.

lucasdemarchi, 2016-12-19:
@LorenzMeier running top on nsh doesn't show such high CPU load. It decreases ~5% reverting the patches I mentioned, but it was already on 50% load.  This is bench testing, not actually flying.

LorenzMeier, 2017-04-30:
Fixed.

closed 2017-04-30 by LorenzMeier