PX4/PX4-Autopilot #6042: Recent PR to "run posix simulator directly from IDE" has broken the Snapdragon build

jywilson, 2016-12-12:
@anton-matosov 

Your last commit (https://github.com/PX4/Firmware/commit/fff493474356f34c32075cc4f67e50d787efa7a2) has broken the Snapdragon Flight board target build:

make eagle_legacy_default

The problem appears to be in your addition of the following call in the function function(px4_add_sitl_app) in file ./src/Firmware/posix/CMakeLists.txt:

FASTRPC_STUB_GEN(../qurt/px4muorb.idl)

This same function is also called in ./src/firmware/qurt/CMakeLists.txt when building the eagle_legacy_default target and causes an error during make file generation:

add_custom_target cannot create target "generate_px4muorb_stubs" because
  another target with the same name already exists.

FASTRPC_STUB_GEN(../qurt/px4muorb.idl) is also called in ./src/firmware/qurt/CMakeLists.txt and somehow by moving FASTRPC_STUB_GEN into the newly created function(px4_add_sitl_app) it has created a conflict. I am not sure why this would be different than the original CMake code which has the same call, but outside the context of the function px4_add_sitl_app().  Perhaps functions treat targets differently.

Can you please resolve this?  Anything you can do to fix this quickly, would be appreciated.  It is blocking the development of code for the Snapdragon platform.

Jim W.

LorenzMeier, 2016-12-12:
It also broke other POSIX flight targets. My bad for not seeing that in the review.

@eyeam3 @anton-matosov We need to make sure to treat SITL and POSIX targets differently. One is simulation, the other are real drones. SITL can depend on POSIX, but POSIX must not depend on SITL.

jywilson, 2016-12-12:
> We need to make sure to treat SITL and POSIX targets differently.  

I agree.  The overlap makes it very confusing to understand what the CMake output will actually produce.

LorenzMeier, 2016-12-12:
Please continue the discussion here: https://github.com/PX4/Firmware/pull/6043. I've fixed on master by reverting.

closed 2016-12-12 by LorenzMeier

anton-matosov, 2016-12-13:
ok. will rework this

eyeam3, 2016-12-13:
Maybe you don't have to. Have a look at #6049. I restructured the file a little, so that the posix targets work. I also tried to keep your modifications but was not able to test.

anton-matosov, 2016-12-13:
Ok. let me checkout your branch