PX4/PX4-Autopilot #6051: HIL_CONTROL message not sent in v1.5.2

mzahana, 2016-12-13:
Hi,

I am developing HIL interface with Pixhawk. I have been using v.1.4.4. To use HIL for quadcopter, I flash the Quadrotor HIL X airframe from QGC. Then, send the HIL sensor and GPS messages to Pixhawk from the Simulator, VREP. Finally, I get back the HIL_CONTROL message from Pixhawk in order to use for control in the simulator. All worked fine in v1.4.4. When I upgraded to v1.5.2, that didn't work. In MAVLink inspector from QGC, I could neither see the HIL_CONTROL nor HIL_ACTUATOR_CONTROL messages. Is there a different way in v1.5.2 to activate HIL mode in order to get these messages?

Thanks.

julianoes, 2016-12-15:
Could it be something about whether it's armed or disarmed? There was this change #5987 .

mzahana, 2016-12-15:
I also tried when the vehicle is armed, but the same issue is there.

mzahana, 2016-12-19:
For the recent version, v1.5.2, do I need to send a command to px4 to tell it to go into HIL mode?

v01d, 2016-12-19:
Maybe related to #5976 ?

mzahana, 2016-12-19:
It is not clear to me what the problem is!

v01d, 2016-12-19:
I guess it is due to this: https://github.com/PX4/Firmware/commit/010c9e937b23988b222de6a1f714b57690f7742f

julianoes, 2016-12-19:
@v01d you're right, we adapted the mavlink message because the previous one was used incorrectly.

mzahana, 2016-12-19:
Does that mean the HIL mode has to be enabled, before the HIL_ACTUAOT_CONTROL is streamed?

v01d, 2016-12-21:
@mzahana By reading the code it would seem that HIL needs to be enabled by setting the bit MAV_MODE_FLAG_HIL_ENABLED in the "set mode" command. Using mavros I'm not sure who should set this bit. I'm guessing that QGC sets it using the HIL config widget. Since V-Rep is not supported, I wonder if QGC still supports setting this mode. Otherwise, it should be possible to do it manually using mavros tools.

LorenzMeier, 2016-12-26:
Closing as it got replaced by HIL_ACTUATOR_CONTROL.

closed 2016-12-26 by LorenzMeier

mzahana, 2016-12-26:
Lorenz: could you please confirm whether HIL_ACTUATOR_CONTORL needs to be activated before it is sent or not? 

It was not sent by Pixhawk either!

LorenzMeier, 2016-12-26:
It's sent if the system is in a HIL config and HIL got enabled.

mzahana, 2016-12-26:
Thanks. Could you please clarify how to enable HIL  given that it is already in HIL config? Is there some MAVLINK command?

mzahana, 2016-12-27:
I could get HIL_ACTUATOR_CONTROL in recent firmware (1.5.1+). The vehicle has to be armed before the topic is published. Thanks.