PX4/PX4-Autopilot #6059: Inspect yaw mixer limiting

LorenzMeier, 2016-12-14:
@Stifael @MaEtUgR I suspect a regression in the yaw mixer limiting. Can you look at that with @priseborough with high prio tomorrow?

priseborough, 2016-12-14:
@LorenzMeier can you elaborate on the reasons behind your suspicion?

I ran a git blame on mixer_multirotor.cpp.  

https://github.com/PX4/Firmware/blame/c4ccfeae0dcd2a16a676196a9194fbcd8af6e78a/src/modules/systemlib/mixer/mixer_multirotor.cpp

The recent changes to support integration limiting were reporting only, were added inside the rate  limiter and did not change the mixing logic as far as I can tell. The most recent functional change before that were in August/ September 2016 and involved addition and relocation of a slew rate limiter. The last time the core mixing algorithms were touched was in 2015.

priseborough, 2016-12-15:
I ran a test on the 250-Quad that shows a net thrust increase with large yaw demands and was able to show in the logs that this was not due to mixing, but motor nonlinearity. Here are the yaw demands and sum of esc demands over a period with large yaw disturbances that cause the height to increase with each correction.

[image]

Here are the esc demands:

[image]

This is hardly surprising - the esc is essentially controlling motor supply voltage proportional to PWM value, static thrust varies with RPM ^2 and therefore the thrust from two motors at 100% PWM is greater than 4 at 50% PWM.

LorenzMeier, 2016-12-18:
@priseborough I'm referring to the poor tracking on yaw commands here:
http://review.px4.io/plot_app?log=fc450a61-29a5-4dbd-8142-bbedffef577c

This might be combined with an improper setup of the vehicle in terms of mixing, CG or alike.

LorenzMeier, 2016-12-18:
It was mixing. We still need to make sure the throttle linearisation is set correctly as well.
https://github.com/PX4/Firmware/pull/6086

LorenzMeier, 2016-12-18:
@tumbili @priseborough I would like to enable the compensation with a conservative and tested default. Right now it is zero: https://github.com/PX4/Firmware/commit/bad2dc80a4458a2f69800cc3029f424fe5e468df

LorenzMeier, 2016-12-20:
@tumbili @Stifael @MaEtUgR @priseborough Log with the linearisation set to 0.65:
http://review.px4.io/plot_app?log=5387f0af-8576-4419-bed2-f799919331bb
http://review.px4.io/plot_app?log=da33e428-8456-4cde-9aff-1452b8d4388a

Pilots like it better. Note that the GPS quality is crappy in these plots, so don't believe the ground trajectory.

kd0aij, 2016-12-20:
What improvements were observed with THR_MDL_FAC = 0.65 and is it possible to see them in the logs?
I believe these flights were on a (stock 3S?) 3DR Iris.

priseborough, 2016-12-21:
I tested master on a 250 quad running a pixracer and did not see any change in esc outputs with adjustment of the THR_MDL_FAC from 0 to 1, when producing full authority yaw demands, so do not think it is working on that board type.

kd0aij, 2016-12-23:
@priseborough Interesting: how did you make the comparisons between ESC outputs for different parameter values? Using log data or additional debug output?

priseborough, 2016-12-23:
I was using log data

LorenzMeier, 2016-12-26:
Ok, so it seems all decent now and we fixed by using the proper mixer.

closed 2016-12-26 by LorenzMeier