PX4/PX4-Autopilot #6062: Ho to Upload Pixhawk firmware

danil2009, 2016-12-15:
Hi Lorenz,
I want to change my Flight control from APM2.5 to Pixhawk of my Quad-copter, but I had have confuse about select a firmware to upload on Pixhawk? my questions is;

    What is the differences the firmware from ardupilot px4 [http://firmware.eu.ardupilot.org/Copter/2016-12/2016-12-14-10:12/PX4-quad/] vs https://github.com/PX4/Firmware/releases. Which one is stable?
    What is the tools to upload firmware from https://github.com/PX4/Firmware/releases to pixhawk?
    Can I use qgroundcontrol ?
    Could you help me please, how to upload firmware step by step (tutorial)
    Thank you

Dida

LorenzMeier, 2016-12-15:
Assuming you want to flash PX4 (since this is the PX4 issue tracker): Download QGroundControl, connect the Autopilot via USB, click on Firmware upgrade in the settings menu, select PX4 click on Upgrade, done!

closed 2016-12-15 by LorenzMeier

danil2009, 2016-12-15:
Ok,  I try as soon. Thank you very much Lorenz

danil2009, 2016-12-15:
What is the differences the firmware from ardupilot px4 [http://firmware.eu.ardupilot.org/Copter/2016-12/2016-12-14-10:12/PX4-quad/] vs https://github.com/PX4/Firmware/releases. Which one is stable?

LorenzMeier, 2016-12-15:
Use QGroundControl. Don't download the software yourself.

danil2009, 2016-12-15:
Oke, Thank you Lorenz