PX4/PX4-Autopilot #6067: [Pixracer] Weird sensors behavior on startup

anton-matosov, 2016-12-15:
Pixracer gets a kind of constant bias on its sensors. I was able to capture it in QGC this time.
https://youtu.be/c7R5lsxxlKw

I have experienced this issue several times already and px4 still allowed to arm it and flipped immediately at takeoff. Unfortunately I don't have logs for this yet. I will try to download logs if I get this issue on the field again.

anton-matosov, 2016-12-15:
```
2000_01_01_03_21_58: [lpe] land init
2000_01_01_03_21_59: [lpe] baro init 67 m std 24 cm
2000_01_01_03_21_59: [lpe] z resume 
2000_01_01_03_21_59: [lpe] tz resume 
2000_01_01_03_22_00: [lpe] land fault,  beta  8.84
2000_01_01_03_22_06: PREFLIGHT FAIL: MAG #1 UNCALIBRATED
2000_01_01_03_22_06: PREFLIGHT FAIL: ACCEL #1 UNCALIBRATED
2000_01_01_03_22_06: PREFLIGHT FAIL: GYRO #1 UNCALIBRATED
2000_01_01_03_22_06: PREFLIGHT FAIL: GYRO SENSORS INCONSISTENT - CHECK
2000_01_01_03_22_17: [lpe] land fault,  beta 76.38
2000_01_01_03_22_19: [lpe] land fault,  beta  9.63
2000_01_01_03_22_35: [lpe] land fault,  beta 119.82
2000_01_01_03_22_36: [lpe] land fault,  beta 11.63
2000_01_01_03_22_53: [lpe] land fault,  beta 145.22
2000_01_01_03_22_54: [lpe] land fault,  beta 14.34
2000_01_01_03_23_02: data link #0 lost
2000_01_01_03_23_11: [lpe] land fault,  beta 126.96
2000_01_01_03_23_12: [lpe] land fault,  beta 13.34
2000_01_01_03_23_28: [lpe] land fault,  beta 172.71
2000_01_01_03_23_29: [lpe] land fault,  beta 11.45
2000_01_01_03_23_46: [lpe] land fault,  beta 227.65
2000_01_01_03_24_04: [lpe] land fault,  beta 258.49
2000_01_01_03_24_22: [lpe] land fault,  beta 401.12
2000_01_01_03_32_53: PREFLIGHT FAIL: ACCEL #1 UNCALIBRATED
2000_01_01_03_32_53: PREFLIGHT FAIL: GYRO #1 UNCALIBRATED
2000_01_01_03_32_53: PREFLIGHT ADVICE: ACCEL SENSORS INCONSISTENT - CH
2000_01_01_03_32_53: PREFLIGHT FAIL: GYRO SENSORS INCONSISTENT - CHECK
2000_01_01_03_32_53: data link #0 regained
2000_01_01_03_33_27: [lpe] land fault,  beta 34678.63
2000_01_01_03_33_38: PREFLIGHT FAIL: GYRO SENSORS INCONSISTENT - CHECK
2000_01_01_03_33_38: CAUTION: Avionics power low:   4.71 Volt
2000_01_01_03_33_38: ARMED by arm/disarm component command
2000_01_01_03_33_38: no RC
2000_01_01_03_33_38: failsafe mode on
2000_01_01_03_33_39: [logger] file: /fs/microsd/log/sess004/log001.ulg
2000_01_01_03_33_46: DISARMED by arm/disarm component command
2000_01_01_03_33_46: failsafe mode off
2000_01_01_03_35_47: data link #0 lost
2000_01_01_03_35_48: PREFLIGHT FAIL: ACCEL #1 UNCALIBRATED
2000_01_01_03_35_48: PREFLIGHT FAIL: GYRO #1 UNCALIBRATED
2000_01_01_03_35_48: PREFLIGHT ADVICE: ACCEL SENSORS INCONSISTENT - CH
2000_01_01_03_35_48: PREFLIGHT FAIL: GYRO SENSORS INCONSISTENT - CHECK
2000_01_01_03_35_48: data link #0 regained
```

[sess004-log001.ulg.zip](https://github.com/PX4/Firmware/files/655938/sess004-log001.ulg.zip)

LorenzMeier, 2016-12-15:
Your sensors are not calibrated. You need to calibrate them. Its possible that some of that got reset in the last upgrade because of an unavoidable calibration data bug fix for Pixracer.

CarlOlsson, 2016-12-18:
We also have strange sensor behaviors on Pixracer sometimes on bootup. After repowering it is usually ok. @priseborough do you think this issue is related to our problems?

LorenzMeier, 2016-12-26:
Is this fixed on master for you both? @CarlOlsson please provide the serial log if you see the next failure.

priseborough, 2016-12-26:
@CarlOlsson Our symptons are an unexplained large offset on the X gyro that occurs intermittently on one board

LorenzMeier, 2016-12-26:
We need more data on this one.

LorenzMeier, 2016-12-26:
I've significantly improved this in fff35fe34b. If it still fails now I would consider that your hardware is bad and got unnoticed so far.

closed 2016-12-26 by LorenzMeier

maciek01, 2017-03-31:
i am experiencing the same issues with 1.5.5 firmware release