PX4/PX4-Autopilot #6070: jMavSim not happy with EKF2 in HITL mode

lovettchris, 2016-12-16:
Is jMavSim supposed to be happy with the new EKF2 ?  I am also having trouble with HITL simulation where jMavSim flies crazy.  If I do this on 1.5.1 it works better:

    param set SYS_MC_EST_GROUP 1

But that doesn't work on 1.5.2 because the old LPE was removed completely.  So if I re-enable it by editing the src/configs/nuttx_px4fmu-v2_default.cmake file so it looks like this:

#
# Estimation modules
#
modules/attitude_estimator_q
#modules/position_estimator_inav
modules/local_position_estimator
#modules/ekf2

Then jMavSim HITL simulation is happy again.  Question is whether this was the expected behavior with the changes in 1.5.2 ?  Does jMavSim need updating to work with EKF2 or am I missing some other setup step ?

LorenzMeier, 2016-12-16:
Hi Chris,

Sorry for the churn in the estimator default adjustments. It likely needs an adjustment on the jMAVSim side (since Gazebo is working). Potentially in the simulated GPS noise accuracy or gyro integrals? What kind of errors are you getting?

-Lorenz

clovett, 2016-12-16:
What I’m seeing is the drone flip upside down and crash, then it keeps on flipping over like a fish out of water…  in other words completely out of control.

tumbili, 2016-12-16:
@lovettchris Try setting the param EKF2_MAG_TYPE to 1. I remember that 3D mag fusion was not working with jmavsim for some reason. Not sure if this is related.

lovettchris, 2016-12-17:
That's a bit better bit still very wobbly and out of control.  I wonder if that pesky HIL_STATE_QUATERNION message is causing problems again in jmavsim... I'll just use the old LPE for now.

closed 2018-02-21 by PX4BuildBot