PX4/PX4-Autopilot #6077: Current Master broken!

tops4u, 2016-12-16:
Tried to install the actual Master on my AUAV-X2 ended up in this:

> ERROR [sensors] Mag #0 failover:  Data timeout!
> Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: hpwork
> sp:     20005c38
> IRQ stack:
>   base: 20005c7c
>   size: 000002e8
> 20005c20: 080d0680 000000b8 080d05ba 20005c38 10001528 0809d4c5 080d0680 000000b8
> 20005c40: 08097b29 00000010 20003f84 00000003 00000000 0809d755 0809d741 080a5c11
> 20005c60: 00000000 20018d90 100014c0 00000006 00000000 0809d6c5 100013b4 00000000
> sp:     10001488
> User stack:
>   base: 10001528
>   size: 0000063c
> 10001480: 80000010 08006363 00000000 00000000 00000002 00000010 00000010 080a5ce7
> 100014a0: 00000002 100007a0 100014c8 080a3239 00000000 00000000 00000000 00000000
> 100014c0: 00000000 080988fd 00000004 00000000 080149d1 00000000 00000000 00000000
> 100014e0: 00000000 20018e14 00000000 00000005 20003e10 00000000 00000000 0809b0ab
> 10001500: 0809b0dd 00000000 00000000 00000000 00000000 0809b0e5 0809b0dd 08098ffb
> 10001520: 00000000 00000000 ffffffff c2b1f913 a593420f 8c2a9e40 000001a0 80000650
> R0: 00000000 00000000 100014c0 76666b19 20018d90 100014c0 00000006 00000000
> R8: 004e4ce9 00000000 00000000 00000000 00000000 10001488 08006379 a0000000
> xPSR: 61000000 BASEPRI: 00000000 CONTROL: 00000000
> EXC_RETURN: ffffffe9
>

LorenzMeier, 2016-12-16:
Thanks for reporting. This is possible as we do a lot of testing on master but do not guarantee its stability at all times: http://dev.px4.io/starting-contributing.html#contributing

@bkueng I think this is super likely the sensors app cleanup / rewrite.

tops4u, 2016-12-17:
I know, just wanted to report.

LorenzMeier, 2016-12-17:
I can reproduce this now:

```
nsh> hmc5883 stop
nsh> Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: hpwork
sp:     20005c38
IRQ stack:
  base: 20005c7c
  size: 000002e8
20005c20: 080d0710 000000b8 080d064a 20005c38 10001528 0809d555 080d0710 000000b8
20005c40: 08097bb9 00000010 20003f84 00000003 00000000 0809d7e5 0809d7d1 080a5ca1
20005c60: 00000000 1000aa30 1000ab00 100014c0 00000000 0809d755 100013a4 00000000
sp:     10001478
User stack:
  base: 10001528
  size: 0000063c
10001460: 40a00000 43d88000 3f7fc826 3efe3013 20000010 08014895 c2b40000 43a58000
10001480: 43d88000 00000000 b1014b01 00005a00 016f8246 00000000 00000000 00000000
100014a0: be1a65fc 3eb168bc 3efe3013 3ff33333 3a9fd80a 421b0000 00010000 feb5005a
100014c0: 000001b1 00000000 080b35c7 1000aa30 00000000 00000001 20003e10 00000000
100014e0: 00000000 00000000 00000000 0801496b 1000aa30 20003e14 1000aa64 0809b13b
10001500: 0809b16d 00000000 00000000 00000000 00000000 0809b175 0809b16d 0809908b
10001520: 00000000 00000000 ffffffff 55c6b6a0 7a817459 83fa2964 000001a0 80000650
R0: 1000aa30 00000001 0000090c 00000000 1000aa30 1000ab00 100014c0 00000000
R8: 0000014b 00000000 0000005a 000001b1 00000000 10001478 0801489f 00000000
xPSR: 00000000 BASEPRI: 00000000 CONTROL: 00000000
EXC_RETURN: ffffffe9
```

LorenzMeier, 2016-12-17:
One more instance with GCC 5.4 (which has a GDB version which doesn't segfault, so I can actually debug it):

```
nsh> INFO  [lib__ecl] EKF alignment complete
hWARN  [commander] Not ready to fly: Sensors not set up correctly
mc5883 stop
nsh> Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: hpwork
sp:     20005c88
IRQ stack:
  base: 20005ccc
  size: 000002e8
20005c80: 10001528 0809cac3 080cfa88 000000b8 0809712d 00000010 20003fd4 00000003
20005ca0: 00000000 0809cd61 0809cd4d 080a51f1 00000000 1000b700 1000b7d0 100014b8
20005cc0: ffffff93 0809ccd1 1000139c 00000000 00000000 00000000 20006ba0 20006c48
sp:     10001470
User stack:
  base: 10001528
  size: 0000063c
10001460: 3f7fc826 3f0dc12b 20000010 08014745 c2da0000 43480000 43f00000 00000000
10001480: e001c800 42826d00 018dac6f 00000000 00000000 00000000 be305808 3e3866c8
100014a0: 3f0dc12b 3ff33333 3a9fd80a 41ae0000 00010000 ff38006d 000001e0 00000001
100014c0: 20003e64 1000b700 00000000 00000005 20003e64 00000000 20003e60 20003998
100014e0: 00000000 0801479f 1000b700 20003e64 1000b734 0809a669 00000000 00000000
10001500: 00000000 00000000 00000000 00000000 00000000 0809a6a9 0809a6a1 08098617
10001520: 00000000 00000000 ffffffff 55c6b6a0 7a857459 83fa2964 000001a0 80000650
R0: 1000b700 00000001 0000062a 445f4d57 1000b700 1000b7d0 100014b8 ffffff93
R8: 00000000 0000006d 000000c8 ffffff38 00000002 10001470 0801474f 445f4d56
xPSR: 01000000 BASEPRI: 00000000 CONTROL: 00000000
EXC_RETURN: ffffffe9
```

LorenzMeier, 2016-12-17:
Binary:
[firmware_nuttx.zip](https://github.com/PX4/Firmware/files/658954/firmware_nuttx.zip)

LorenzMeier, 2016-12-17:
```
Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: hpwork
sp:     20005c88
IRQ stack:
  base: 20005ccc
  size: 000002e8
20005c80: 10001528 0809cb23 080cfb04 000000b8 0809718d 00000010 20003fd4 00000003
20005ca0: 00000000 0809cdc1 0809cdad 080a5251 00000000 20012540 100014b0 00000006
20005cc0: 00000000 0809cd31 100013a4 00000000 00000000 00000000 20006ba0 20006c48
sp:     10001478
User stack:
  base: 10001528
  size: 0000063c
10001460: 40a00000 00000000 34000000 3ca3d70a 80000010 08006365 00000000 00000000
10001480: 00000002 00000010 00000010 080a5327 00000002 100007a0 100014b8 080a28d9
100014a0: 00000000 00000000 00000000 00000000 00000000 08097f69 00000004 00000000
100014c0: 08014819 1000aa64 00000000 00000000 00000000 200125c4 00000000 00000005
100014e0: 20003e64 00000000 20003e60 20003998 00000000 0809a6c9 00000000 00000000
10001500: 00000000 00000000 00000000 00000000 00000000 0809a709 0809a701 08098677
10001520: 00000000 00000000 ffffffff 57c6b621 7a857459 81fa2964 000001a0 80000650
R0: 00000000 00000000 100014b0 3a3435c6 20012540 100014b0 00000006 00000000
R8: 0038c449 00000000 20003998 00000000 00000000 10001478 0800637b 3a3435c6
xPSR: 60000000 BASEPRI: 00000000 CONTROL: 00000000
EXC_RETURN: ffffffe9
```

LorenzMeier, 2016-12-17:
With more FDs and guards in HMC5883 still happening, looks like stack smashing:

```
nsh> hmc5883 stop
nsh> Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: hpwork
sp:     20005c88
IRQ stack:
  base: 20005ccc
  size: 000002e8
20005c80: 10001648 0809cae3 080cfab8 000000b8 0809714d 00000010 20003fd4 00000003
20005ca0: 00000000 0809cd81 0809cd6d 080a5211 00000000 1000aef0 1000afc0 100015d8
20005cc0: ffffff97 0809ccf1 100014bc 00000000 00000000 00000000 20006c90 20006d38
sp:     10001590
User stack:
  base: 10001648
  size: 0000063c
10001580: 3f7fc826 3f050542 20000010 0801474d c2d20000 433f0000 43e20000 00000010
100015a0: c401bf00 080a6900 02aa3e4d 00000000 00000000 00000000 be2bb948 3e2cb186
100015c0: 3f050542 3ff33333 3a9fd80a 42120000 00010000 ff410069 080101c4 10009094
100015e0: 00000000 1000aef0 00000000 00000005 20003e64 00000000 20003e60 20003998
10001600: 00000000 080147a7 1000aef0 20003e64 1000af24 0809a689 00000000 00000000
10001620: 00000000 00000000 00000000 00000000 00000000 0809a6c9 0809a6c1 08098637
10001640: 00000000 00000000 ffffffff aa60f54b f8f97c6b ed453afa 000001a0 80000650
R0: 1000aef0 00000001 0000120e 00000000 1000aef0 1000afc0 100015d8 ffffff97
R8: 00000000 00000069 000000bf ffffff41 00000002 10001590 08014757 00000000
xPSR: 00000000 BASEPRI: 00000000 CONTROL: 00000000
EXC_RETURN: ffffffe9
```

LorenzMeier, 2016-12-17:
Turns out that my reproduction efforts of the original issue had its own flaw. I need to test your stack dump against the binary from the CI server. @davids5 Would be good if you can also have a look.

I did fix a number of things in the process for the case of a single mag in the system, all that is here:
https://github.com/PX4/Firmware/pull/6081

I need to verify it fixes this original issue. @simonegu @julianoes Please keep an eye on this. We need to be absolutely certain we fixed it before closing it.

tops4u, 2016-12-17:
Just to be sure. My System shuld habe two Compass. One (or more?) from the AUAV X2 and one from the Drotek GPS via I2C.

LorenzMeier, 2016-12-17:
If you have a shell connected you can run `sensors status`. The output of that would be helpful. Can you also test if it reliably crashes if you disconnect the GPS? Please try 10 or 20 times if it doesn't reproduce immediately. That will be super helpful for us, thanks!

tops4u, 2016-12-17:
What Firmware Level shall I test this on? 1.5.2? Master did not come up as stated above. I also get the Error Ststus from the LED Signals on the FC.

LorenzMeier, 2016-12-17:
Can you try current master again? Does it crash this way absolutely every boot?

LorenzMeier, 2016-12-17:
Don't change anything, just try to boot again.

tops4u, 2016-12-17:
So I did some testing. My Answer may give you some headache - sorry for that.

First I want back to Master (as I flashed 1.5.2 yesterday). This went smooth also the 1. restart right after the Upgrade was ok. Then the 1st Power-cycling did it and I got the error. I did cycled Power many times afterwards, but was not able to reproduce. Also unplugging the external Compass (#0) never caused this Problem - except that I got sometimes many Messages that the State went back to -1 but no Data Timeout Switchover Message (lost?) on the console. 

So I thought may be this has something to do with the first power-cycle after a restart. So I went back to Stable and to Master again. This Time also after Power Cycle nothing happened. But on the 2nd Power Cycle and after about the 6th. 

So to me this seems rather random, and so far I could not detect any pattern - at least I could not provoke this by unplugging the I2C bus to the external Compass.

Anything else I could try?

LorenzMeier, 2016-12-18:
Yes. Please flash this binary and see if you can reproduce it. If you can, please share the stack dump.

[nuttx-px4fmu-v2-default.px4.zip](https://github.com/PX4/Firmware/files/659403/nuttx-px4fmu-v2-default.px4.zip)

@CarlOlsson You mentioned seeing boot issues on Pixracer - could you please connect the shell and try to reproduce it?

tops4u, 2016-12-18:
Looks like it is fixed. I did 3 times Upgrades from _1.5.2 -> Your File_ and about 25 Power Cycles as well as about 10-15x I2C disconnects. The Problem did not reoccur. 

Just one minor note: In QGC 1.5.2 is identified as 1.5.1dev while your File is identified as 1.5.2.

kd0aij, 2016-12-18:
latest master on PixRacer:
HW arch: PX4FMU_V4
FW git-hash: aa8e43b26992aa1647d945505f33c1d5f150cae0
FW version: pwm_trim_params-107-gaa8e43b (0.0.0 0), 0
OS version: 6.27 (100796415)
Build datetime: Dec 18 2016 14:16:47
Toolchain: 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
MCU: STM32F42x, rev. 3
UID: 1D0031:3235510C:32393433 

Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: hpwork
sp:     20006f48
IRQ stack:
  base: 20006f8c
  size: 000002e8
20006f40: 10001528 080de275 08120d1c 000000b8 080d8a8d 00000010 2000471c 00000003
20006f60: 00000000 080de505 080de4f1 080e6cb5 00000000 10007300 10001488 00000006
20006f80: 00000000 080de475 10001374 00000000 10000010 00000001 00000000 00000000
sp:     10001448
User stack:
  base: 10001528
  size: 0000063c
10001440: 80000010 080065a1 a0829b73 3fc85a7a 00000000 00000000 00000000 00000000
10001460: 00000000 00000000 00000000 00000000 00000000 437a0000 00000000 00000000
10001480: 00000000 00000000 00000000 00000000 00000002 00000010 00000010 080e6d8b
100014a0: 00000002 100007a0 100014c8 080e4035 2000428c 00000000 2000428c 00000000
100014c0: 00000000 080d989f 00000004 00000000 0800e061 00000000 00000000 10007384
100014e0: 00000000 00000005 200045a8 00000000 00000000 00000000 00000000 080dc12b
10001500: 080dc15d 00000000 00000000 00000000 00000000 080dc165 080dc15d 080d9f9f
10001520: 00000000 00000000 ffffffff 5e0f493d 96601534 b1fb14ef 000001a0 80000650
R0: 00000000 00000000 10001488 200139ec 10007300 10001488 00000006 00000000
R8: 0048d64a 00000000 00000000 00000000 00000000 10001448 080065b7 200139ec
xPSR: 60000000 BASEPRI: 00000000 CONTROL: 00000000
EXC_RETURN: ffffffe9

LorenzMeier, 2016-12-18:
Can you please share the elf file? (build_px4fmu-v4_default/src/firmware/nuttx/firmware_nuttx).

LorenzMeier, 2016-12-18:
Would be critical to get this. I can fix it with that.

kd0aij, 2016-12-18:
oops, had just finished rebuilding with gcc 5.4; no hardfaults so far
Will regen the 4.9.3 elf file next.

Toolchain: 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]

kd0aij, 2016-12-18:
@LorenzMeier emailed a link to elf binary

LorenzMeier, 2016-12-18:
Initial work shows this is likely the FMU driver. Not sure yet how.

LorenzMeier, 2016-12-18:
@kd0aij aa8e43b26992aa1647d945505f33c1d5f150cae0 is not in tree - where is that commit?

kd0aij, 2016-12-19:
BTW, I haven't been able to reproduce the hardfault with either 4.9.3 or 5.4 builds

kd0aij, 2016-12-19:
I bet the different hash was due to an empty rebase; I changed my git config and tested the editor by doing a rebase with no changes.

LorenzMeier, 2016-12-19:
@bkueng Issue near line 1177 near fmu.cpp:

```C++
if (_to_mixer_status == nullptr) {
				/* publish limits */
				int instance = _class_instance;
				_to_mixer_status = orb_advertise_multi(ORB_ID(multirotor_motor_limits), &multirotor_motor_limits, &instance,
								       ORB_PRIO_DEFAULT);

			} else {
				orb_publish(ORB_ID(multirotor_motor_limits), _to_mixer_status, &multirotor_motor_limits);

			}
```

bkueng, 2016-12-19:
I don't see anything obvious, except that ` = _class_instance;` is not needed, since instance is an output parameter for `orb_advertise_multi`.

LorenzMeier, 2016-12-19:
Fixed here: https://github.com/PX4/Firmware/commit/dfed9a3f13ada3cbf6aa7123649c702f1ad1c566 - thanks for all the reporting!

closed 2016-12-22 by LorenzMeier

FantasyJXF, 2018-08-19:
```
Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: hpwork
sp:     20008d38
IRQ stack:
  base: 20008d7c
  size: 000002e8
20008d20: 080a98b8 000000b8 080a97f2 20008d38 10001528 080778db 080a98b8 000000b8
20008d40: 080721f1 00000010 20006510 00000003 00000000 08077b79 08077b65 0808181d
20008d60: 00000000 00000000 100014bc 0001ae18 200063cc 08077ae9 100013bc 00000000
sp:     10001490
User stack:
  base: 10001528
  size: 0000063c
10001480: 358637bd 3e5974a3 20000011 08075a89 49404b90 2000c5a0 00000000 000c04b9
100014a0: 00000000 00000000 2000639c 20005ed4 00000000 08025579 2000c5a0 000c04b9
100014c0: 00000000 49404b90 2000c5a0 00000000 000c04b9 080255f7 2000c5a0 00000005
100014e0: 200063a0 08025631 2000c5a0 200063a0 2000c5a4 08075749 00000000 00000000
10001500: 00000000 00000000 00000000 00000000 00000000 08075789 08075781 080736df
10001520: 00000000 00000000 ffffffff 97587f7a 9b132311 d7e763bb 000001a0 80000650
R0: 00000000 2002fff8 f780fad0 1783913c 00000000 100014bc 0001ae18 200063cc
R8: f7814b74 200063b4 f780fad0 00000008 00000000 10001490 08072f3d 08075a92
xPSR: 81000000 BASEPRI: 00000000 CONTROL: 00000000
EXC_RETURN: ffffffe9
```

I met the similar problem, can anyone help?

dagar, 2018-08-19:
@FantasyJXF can you open a new issue? Please include the exact firmware version.

FantasyJXF, 2018-08-21:
@dagar Thanks Daniel, I use `arm-eabi-none-gdb` to debug the Firmware.
It's because the stack exceed which caused the bug.

Fixed it .

And I'm developing based on Release 1.5.5.