PX4/PX4-Autopilot #6082: Enable -Weffc++ on more modules

LorenzMeier, 2016-12-17:
@bkueng I found initialization issues and I think we need to enable `-Weffc++` on more modules and eventually for the whole codebase. That is tedious but tends to fine tons of corner cases in initialization.

@julianoes You might want to consider enabling it for the position controller and fix anything it finds. Your re-ordering might have exposed exactly this sort of issue.

dagar, 2017-01-30:
There are some good clang-tidy checks we can enable to really force initialization everywhere. We could even do the bulk of the initial pass automatically. This would be c++11 initialization.

	modernize-use-default-member-init
	readability-redundant-member-init
	cppcoreguidelines-pro-type-member-init
	cppcoreguidelines-interfaces-global-init
	modernize-use-equals-default
	modernize-use-equals-delete
	readability-deleted-default

dagar, 2017-02-05:
@LorenzMeier @bkueng here's a pass of enabling those clang-tidy initialization and modernize checks - https://github.com/PX4/Firmware/compare/master...dagar:c++11_init

There's still a mix of old initializers and new, but so far it looks like it caught everything. They still could benefit from manual cleanup. Moving old c++ initializers into the class, dropping unnecessary memsets, moving default constructors/destructors into the header.

Any suggestions for bringing this in? I could split it into a commit per module in one PR, or do it gradually over time.

LorenzMeier, 2017-04-10:
Using clang-tidy instead.

closed 2017-04-10 by LorenzMeier