PX4/PX4-Autopilot #6089: Ensure all CI systems run on arm-none-eabi-gcc 5.4 q3_2016

LorenzMeier, 2016-12-18:
@dagar Since we have now fault logging and generally good in-system instrumentation we should make sure all our CI systems are on GCC 5.4 and we offer instructions for Linux and Mac OS how to install that exact version.

dagar, 2016-12-18:
Will do once I'm back next week.

dagar, 2016-12-26:
Updating to gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2

Should we do anything to keep older versions working?

dagar, 2016-12-27:
Updated to 5.4 here - https://github.com/PX4/Firmware/pull/6183

dagar, 2017-01-02:
@LorenzMeier builds have been updated to 5.4, but we should probably clarify what's officially supported and if we want to make an effort to keep older GCC working.

closed 2017-01-02 by dagar

mcharleb, 2017-01-02:
@LorenzMeier My current work is on a Yocto Morty based build using GCC 6.2

LorenzMeier, 2017-01-02:
Yep, but you're not building NuttX with that.