PX4/PX4-Autopilot #6091: local_position freeze with updated VICON pose using LPE

weiweikong, 2016-12-18:
I am using the most updated master Firmware, and tried to control the UAV indoor with VICON system. The VICON data sent through WiFi to Odroid and forwarded to PX4 by `/mavros/mocap/pose` or `/mavros/vision_pose/pose`. But the `local_position/local` updated around (0, 0, 0) even though the vision estimated data was updating correctly, monitoring from QGC, when we were moving the UAV randomly. 

According to issue #5945 , it seems that the LPE has worked fine with VICON system. But we got the large errrors between the `local_pose/pose` with `/mavros/vision_pose/pose`.  Also, according to issue #5102, there should be some LPE related paramters, such as
```
x   LPE_INTEGRATE [69,155] : 1
x   LPE_FLW_OFF_Z [70,156] : 0.0000
x   LPE_FLW_XY [71,157] : 0.0100
x   LPE_FLW_QMIN [72,158] : 75
x   LPE_SNR_Z [73,159] : 0.0500
...
```
we could not find those parameters in QGC.

Questions:
1. How should we set the parameters correctly to enable LPE working with VICON or there are no more parameters for us to configure LPE?
2. For `/mavros/mocap/pose` or `/mavros/vision_pose/pose`, which is the best choice to send the VICION data to PX4.

Configurations about my testing platform:
### PX4
- Current Master branch
- Default LPE

### Odroid XU4
- Telem2 with UART 921600
- Lubuntu 14.04
- ROS Jade
- Current MAVROS Branch

### Ground Staion
- Connect with Odroid XU4 using WiFi
- Ubuntu 14.04
- ROS Jade
- ETH Vicion Bridge

LorenzMeier, 2016-12-18:
LPE is only supported on `px4fmu-v3_default` and `px4fmu-v4_default` build targets. If you try load it with the Pixhawk default target `px4fmu-v2_default` you need to enable it `cmake/px4fmu-v2_default` config.

closed 2016-12-18 by LorenzMeier

weiweikong, 2016-12-19:
@LorenzMeier Thanks with the quick reply. I tested the LPE with `px4fmu-v2_default` version, and sent the VICON data to `mocap/pose`, following are the results 
http://logs.uaventure.com/view/HU5iw2K7nyTHCaYbf4sKsk
It seems that the position control is not quite stable(see pictures of local position), when the set position for XY was fixed and we controlled the Z axis through the keyboard.

I already disable the GPS and Vision in LPE parameters and set `PN_V=0.8`.  Is this related to local position control PID parameters?  Would it be possible to share more experiences?

ecmnet, 2016-12-19:
@weiweikong Your acceleration noise seems to be very high. I get quite good LPE results using vision, if the acceleration noise is below 0.1g in all axes.

mhkabir, 2016-12-19:
You can maybe disable accel integration if the accel measurements are bad. It's a define somewhere in LPE.

weiweikong, 2016-12-19:
@ecmnet Should I send VICION data to `vision_pose/pose`? As you mentioned vision, do you mean the optical flow sensor?

weiweikong, 2016-12-19:
@mhkabir Do you mean the `LPE_ACC_XY` and `LPE_ACC_Z` parameters?

weiweikong, 2016-12-19:
@ecmnet You are right that my PX4 was broken which caused the vibration in Z axis. I changed a new one, however, there is still an overshoot in axises, especially in  Z-axis
[image]
As the figure shown, after the overshoot, the UAV could fix near the setting position and the whole log file is http://logs.uaventure.com/view/jHMvjd93pZWM7XMFizwcbG.
My VICION coordinate is ENU.

zzzzzq, 2016-12-25:
@ecmnet 
Hello!
Do you mean accelerate noise density? And I wonder what kind of coordinate system the vision should be? Is it ENU or what?
Thank you!

ecmnet, 2016-12-25:
@zzzzzq Vision is not optical Flow, but vision based position and attitude estimation. I've played around with LPE_ACC_XY and LPE_ACC_Z a bit to get proper results, but acceleration noise was one of the driving factors when it comes to accuracy.