PX4/PX4-Autopilot #6092: Camera trigger now working on pins 5,6

tumbili, 2016-12-19:
I've been testing the camera trigger and haven't got it working on Pins 5&6 of the FMU (TRIG_PINS = 56). I've also tried the setting TRIG_PINS = 34 which works fine.
I'm wondering if there was a bug introduced somewhere because the default pins are in fact 5,6.
@birchera Do you guys have the same issue?

tumbili, 2016-12-19:
Tested on master

mhkabir, 2016-12-19:
Does it work if you stop the trigger app and restart it again via the console? (with pins 5,6)

tumbili, 2016-12-19:
@mhkabir No, I tried all possible things.

tumbili, 2016-12-19:
@mhkabir On which pins have you been using it?

tumbili, 2016-12-19:
BTW, this is using the PWM interface (TRIG_INTERFACE = 2)

mhkabir, 2016-12-19:
I've been using it on 1,2 in GPIO mode. I had previously run into an issue where it would clash with the fmu driver and cause issues. Restarting the trigger app, or starting fmu in mode_gpio beforehand worked. For 5,6 the fmu driver should start in mode_pwm4 leaving 5,6 free. Cross check rcS just in case it isn't for you.

birchera, 2016-12-19:
It works for us. As @mhkabir mentioned you need to start fmu driver in pwm4 mode. Since it runs on 3&4 for you it is likely an issue related to that.

tumbili, 2016-12-19:
@birchera I've checked that as well, FMU is started in mode pwm4. I even killed fmu and cam trigger and started them manually again, same story.

birchera, 2016-12-19:
When you trigger on 3&4 I think you overwrite commands from the fmu driver and they might interfere. I remember we had some issues setting it up differently, but can't quite remember. Let me quickly dive into some old PRs.

tumbili, 2016-12-19:
@birchera Thanks, that would be great!

birchera, 2016-12-19:
We still use the exact same setup as on current [master](https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L397).

What hardware do you use? We have it running on a pixracer, so v4.

tumbili, 2016-12-19:
@birchera I'm trying with pixhawk

tumbili, 2016-12-19:
@birchera Ok, I got one step further. The reason why it worked on pins 34 was because the fmu was put into mode pwm4. In short, the camera trigger only works on those pins which are also configured by the fmu. So now I started fmu in mode pwm (all 6 outputs) and then I was able to use the camera trigger on pins 56. However, I don't think this is the way it was designed to work, right?

birchera, 2016-12-19:
> When you trigger on 3&4 I think you overwrite commands from the fmu driver and they might interfere.

No I don't think that is how it should be. But when we had the pixhawk, we always used the IO pins to trigger. We never tested the fmu ones. Would that be possible for you? With the IO present, some things are set up differently, so it is hard to guess what goes wrong.

tumbili, 2016-12-19:
@birchera How are you using the IO pins for triggering? I wouldn't even know what setting that would be and the documentation on the dev guide only talks about AUX pins.

tumbili, 2016-12-19:
@LorenzMeier Do you as the unquestioned FMU expert have a shotgun idea about what might be going wrong?
To put it in a nutshell: Triggering via the camera trigger app only works if the pins used for triggering are included in the bitmask used by the FMU.
Is there something the FMU does which is not done in the camera trigger app which is needed to output pwm to pins?

birchera, 2016-12-19:
Oh, I meant the AUX pins. Sorry, quite some time I worked with the pixhawk.

mhkabir, 2017-04-03:
Closing in favour of https://github.com/PX4/Firmware/issues/6961.

closed 2017-04-03 by mhkabir