PX4/PX4-Autopilot #6095: Improvement: Backtrack RTL

tops4u, 2016-12-19:
Apparently DJI has released a functionality, where RTL does not simply climb to a specific Height (Above Starting Point) and then return straight to the starting Point but go backward the flightpath which led to this point where RTL is issued. 

Would it be possible to introduce such a functionality ie. by logging the current Flightpath since Arming backwards by logging Waypoints in a specified distance appart (i.e. every 10m). 

This could be particulary handy when flying Missions in Regions where there are Canyons/Forests in uneven terrain, Urban Areas, etc without US/LIDAR.

The logic should also decide if a direct Path should be choosen by analysing the current Power/Battery Situation.

LorenzMeier, 2016-12-19:
There is a pending pull request in the pull request queue which you're welcome to contribute to. Please comment there.

closed 2016-12-19 by LorenzMeier

tops4u, 2016-12-19:
Ok. Sorry I did not find it under the Name "Graph RTL".  Apparently it is almost done except the decision if a direct aproach should be taken when considering Battary/Power requirements.