PX4/PX4-Autopilot #6096: Parameter Binary Search

stephanbro, 2016-12-19:
Based on my current understanding of the parameter search, when doing a parameter find a linear search is done through the parameters (https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param.c#L272). Building the SITL, I counted 917 parameters which could go from a worst case of 917 comparisons to log(917) = ~10 comparisons if the parameters were sorted alphabetically.

Additionally for changed parameters, the utarray doesn't currently use bsearch because it isn't supported in NuttX according to this comment: (https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param.c#L230). The main repo for NuttX has an implementation of bsearch that could be backported for PX4s NuttX found here: https://bitbucket.org/nuttx/nuttx/src/097f09cb0208f246e4c20a79bc459b78bc0f8ed2/libc/stdlib/lib_bsearch.c?at=master&fileviewer=file-view-default

My thoughts for implementation would be to sort the parameters post-xml generation for the .c and .h file generation, modify the param internal sort, to use a binary search, and backport the bsearch in NuttX for the utarray.

Any thoughts or feedback?

bkueng, 2016-12-23:
Sounds good! Note that we just merged a large NuttX upgrade which includes `lib_bsearch`.

jgoppert, 2016-12-23:
@stephanbro @bkueng  I think we should consider using a hash function so we get linear search time. I've already got a string hashing library in this PR: https://github.com/PX4/Firmware/blob/f15318ab842186d603bf73ed3a03e3620137a6a8/src/modules/iekf/ros/ros.cpp#L65. 

It also means we can duplicate the ros parameter system, which is what I'm using it for.

stephanbro, 2016-12-23:
@jgoppert In your implementation of the hash table, you are using a linked list with 128 buckets which leads to a worst case of ~7-8 comparisons with 917 parameters. That's pretty comparable to the binary search comparisons of ~10 without the added memory cost. I guess it just comes down to a tradeoff between the two.

Additionally when generating the parameters, the handle of the parameter will be known at compile time and could be used to create some `#define`'s to prevent using a string comparison for getting the parameter handle internally. The string search would then only be used when mavlink calls param find.

LorenzMeier, 2016-12-23:
I think a binary search is a quite sensible next step, in particular if there is a suitable implementation available.

jgoppert, 2016-12-23:
@stephanbro The hash function is replaceable. You would want to just want to set the table size to 1024. For what I'm doing the 128 works well since I'm just using it for uORB subscriptions. The code is from here: https://github.com/aozturk/HashMap/tree/master/src, I have an issue up about him adding some kind of license: https://github.com/aozturk/HashMap/issues/2

stephanbro, 2017-01-13:
This has been implemented in #6254, so I'm closing it.

closed 2017-01-13 by stephanbro