PX4/PX4-Autopilot #6098: sf1xx problem

lovettchris, 2016-12-19:
There is an unconditional "sf1xx start" in rc.sensors, line 314, then there's another sf1xx start in rcS line 659 that is conditional on a non-zero setting of SENS_EN_SF1XX.  So if you set this to something other than zero this second start fails and causes rcS to abort, which results in no mavlink USB connection :-)  We should only have one of these, it makese sense to live in rc.sensors, and the  check for non-zero SENS_EN_SF1XX also makes sense... so I'd recommend we move the rcS code to rc.sensors...

LorenzMeier, 2016-12-19:
Fixed in 4b99a8f, thanks!

closed 2016-12-19 by LorenzMeier

lovettchris, 2016-12-19:
perfect, thanks!  I see sensor stuff creeping into rcS, what is the guidance on when to use rc.sensors and when not to ?

mzahana, 2017-01-16:
can this be merged into a stable version that can downloaded by QGC directly?