PX4/PX4-Autopilot #6106: Short Unstable Period for Quadrotor After Switching to OFFBOARD Mode

weiweikong, 2016-12-20:
## Status
We are trying to control the position of rotors using OFFBOARD mode with MAVROS by topic `/mavros/local_setpoint ` based on VICON data.
- Firstly, we armed the quadrotor manually by RC on the ground.
- Then, we switch the MANUAL mode to OFFBOARD mode. The quadrotor maneuvered aggressively for the short period as shown in the bellowing picture with blue rectangle.
[image]
- After this short period, the UAV could control normally as shown in the flowing curves.

## Questions
- Why there was an unstable period after switching to OFFBOARD? It seems that the LPE module works fine after the short unstable period?
- After zooming into the detail, as shown in the following picture,  

[image]

each circle denotes the data with timestamp recorded by PX4. The switching signal triggered around 501.4s as shown in orange color. We send the fix setposition point which read from the `mavros/local_position` at the mode switching moment. But the setting point keeped stable around 502s, which has 0.6s delay, as shown in blue circle. We are not sure, whether this delay caused the short unstable peroid?

## Platform Configuration
- Ubuntu 14.04
- Pixhawk 1
- Current Master Firmware Version with LPE module 
- VICON using ENU coordinate
 

- ETH VICON_Bridge
- The original log is http://logs.uaventure.com/view/PTMjRcANNg6y5fex4SU8Aj

weiweikong, 2016-12-22:
It seems that there is a minimum takeoff speed in file `mc_pos_control_main.cpp`, https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp.
```
/* special thrust setpoint generation for takeoff from ground */
	if (_pos_sp_triplet.current.valid
	    && _pos_sp_triplet.current.type == position_setpoint_s::SETPOINT_TYPE_TAKEOFF
	    && _control_mode.flag_armed) {

		// check if we are not already in air.
		// if yes then we don't need a jumped takeoff anymore
		if (!_takeoff_jumped && !_vehicle_land_detected.landed && fabsf(_takeoff_thrust_sp) < FLT_EPSILON) {
			_takeoff_jumped = true;
		}

		if (!_takeoff_jumped) {
			// ramp thrust setpoint up
			if (_vel(2) > -(_params.tko_speed / 2.0f)) {
				_takeoff_thrust_sp += 0.5f * dt;
				_vel_sp.zero();
				_vel_prev.zero();

			} else {
				// copter has reached our takeoff speed. split the thrust setpoint up
				// into an integral part and into a P part
				_thrust_int(2) = _takeoff_thrust_sp - _params.vel_p(2) * fabsf(_vel(2));
				_thrust_int(2) = -math::constrain(_thrust_int(2), _params.thr_min, _params.thr_max);
				_vel_sp_prev(2) = -_params.tko_speed;
				_takeoff_jumped = true;
				_reset_int_z = false;
			}
		}

		if (_takeoff_jumped) {
			_vel_sp(2) = -_params.tko_speed;
		}

	}
```
So the quadrotor should take off in an aggressive maneuver ignoring the OFFBOARD set points until it reaches the `tko_speed`. Is that right?

weiweikong, 2016-12-27:
@Stifael In MAVROS issue, https://github.com/mavlink/mavros/issues/610#issuecomment-269236188, @cookz2 and we seem to find the similar scenario that the vehicle **has to reach the** `tko_speed`, the minimum taking off speed **before** it maneuvered correctly with the local position control in OFFBOARD mode.  

Is this right? or is there any advice related to the OFFBOARD control? Flying in the air then switch the vehicle to OFFBOARD is better or ARM and switching the mode all through the MAVROS is recommanded?

Stifael, 2016-12-28:
@weiweikong: 
I tried to reproduce your results. Unfortunately, I can only test it in SITL, since I dont have the setup at hand. 
Anyway, the way I did it was to to send the offboard messages continuously as opposed to sending them with the manual-to-offboard switch. In theory, that should not make a difference, but I guess it is worth to try, since I did not encounter your issue of a an aggressive maneuver. 
 
Below are the plots: 
I basically set the offboard setpoint to (x=0,y=0,z=10) relative to the initial position. then I switched between offboard and manual, where the the switch is apparent when looking at the reference setpoint with offboard mode when the line is dotted and manual mode otherwise. 
The switching is best apparent from the z-plot.

[image]

[image]

[image]

Stifael, 2016-12-28:
about the minimum tko_speed: 
the special takeoff is supposed to happen only if the position setpoint is a takeoff, if should not be the case if you are already in air.
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1131-L1132

weiweikong, 2017-01-01:
@Stifael Thank you for your testing. Recently, I was also tested my platform and it worked even worse. The vehicle could not control correctly in OFFBOARD mode. The control procedure were
- ARM the rotors with RC manually.
- Get the current `local_position/local` and send the data to `setpoint_position/pose` by MAVROS and the VICON data is transferred to `mocap/pose` (We did the msg format transfer) whose coordinate is ENU (x is pointing to X-axis, y is forward and z is upward).
- After checking the `setpoint_position/pose` status, switched the mode to OFFBOARD by RC manually.

The quadrotor maneuvered strangely
- although we set the `setpoint_position/pose` is the current pose, the vehicle still took off directly.
- Different from the previous test, the vehicle flew like a Toilet Bowl Effect.

The log file is http://logs.uaventure.com/view/K7pJNTHtvxZZBUXrn5Qiz4 and we also record the video for analysis, https://www.youtube.com/watch?v=CklqJmYQlFw.

Would it be possible to share more experience about indoor flight testing with VICON? Thanks.

Stifael, 2017-01-03:
Just to be sure that we are talking about the same thing when talking about the ENU frame attached to the vehicle: 
x is pointing forward (rotating around x is rolling)
y is pointing to the left (rotating around y is pitching)
z is point upward (rotating around z is yawing).
This is opposite of what you said, that y is pointing forward and not sure what you mean with x pointing in X-axis. 
I also was wondering to set the setpoint_position/pose to a fixed value just to see if the vehicle goes there, or is that what you did when you set the setpoint to the current pose or did you reset the setpoint to the newest current position?

weiweikong, 2017-01-03:
Sorry for my typo, to make sure the frame, please refer the following diagram

[image]

### PX4
- The coordinate is NED, and the local position data in the log file are in this coordinate.
- MAVROS transfer the ROS-ENU to PX4-NED automatically.

### ROS
- Without VICON, the topic `/mavros/local_position/pose` publishes the UAV position in ENU, whose original point (0, 0, 0) is the position where the PX4 turns on.
- With VICON, whose coordinate is ENU as shown in the diagram, the topic `/mavros/mocap/tf` publishes the VICON data to PX4, so the data `/mavors/local_position/pose` should be almost the same with `/mavros/mocap/tf`. We used `rostopic echo` and make sure that these to topics are have the similar position.

### Experiment Configuration
- We set the vehicle on the ground, and when the  `/mavors/local_position/pose`  was stable, we save the current position and send the fixed point to `/mavros/setpoint_position/local` before switching into OFFBOARD mode.
- Considering above situation, we consider the quadrotor should be on the ground, although the speed of rotors are faster than idle status, or take off for a short distance above the ground.
- But it maneuvered as the video shown.

### Questions
- Should we set VICON ENU perfectly aligned the real EAST and NORTH? In the current situation, we set the VICON ENU coordinate arbitrarily which did not consider the X-axis of VICON ENU coordinate is pointing to real EAST.
- Considering the above setting, should we set `ATT_EXT_HDG_M = 2` as recommended in http://dev.px4.io/external-position.html?

Stifael, 2017-01-03:
@weiweikong:
Yes, the frames look correct. Unfortunately I never set up the vicon system, so I cannot really give you advice from my own experience. But I just read the px4 devguide about motion capture systems with px4 where it says that ATT_EXT_HGD_M has to be set if you want to set the heading externally, so I think you should set it to 2. Also I think you should send data to /mavros/mocap/pose. 
@mhkabir: I have seen you have commented in the past about mocap system. Do you have any suggestions?

mhkabir, 2017-01-03:
If your VICON coordinate system isn't aligned with real ENU, you must set ATT_EXT_HDG_M = 2.

weiweikong, 2017-01-06:
@mhkabir I tried to set `ATT_EXT_HDG_M = 2` and the VICON coordinate system is not aligned with real ENU, but the vehicle still flew in turning circle.

mhkabir, 2017-01-06:
Are you sending the data to /mavros/mocap/pose ? This "turning circle", this happens in Position Control, yes?

weiweikong, 2017-01-06:
@mhkabir Yes, I publish the VICON data to `/mavros/mocap/pose` and we were under OFFBOARD mode with Positon Control. (Firmware Release 1.5.1 Version with LPE module)

mhkabir, 2017-01-06:
Don't use offboard. I'm trying to eliminate all other sources of issues.

Just try to stably hover in Position control mode (POSCTL) and send a log.

weiweikong, 2017-01-10:
@mhkabir Sorry for my late testing. The log files are http://logs.uaventure.com/view/x6jyxefFjqWqpqiwWQX9B3 and http://logs.uaventure.com/view/qVQnmmjgXvNnpTLt7i8M4e. One of the related video is at https://www.youtube.com/watch?v=VPStdRqAzUU.

mhkabir, 2017-01-10:
@Stifael Any idea about these crazy setpoint jumps on @weiweikong's logs ? 

[image]

[image]

Stifael, 2017-01-10:
Looks rather strange and I dont think that these plots makes a lot of sense, since the estimation stays perfectly at zero (at least pos y and dont know about x bc of scaling but since it was indoor test it is most likely zero as well).Also when looking at the other plot, the position tracking is really bad. Are these logs created with the exact same setup?

mhkabir, 2017-01-10:
Actually, the auto-scaling because of those transients makes it seem that they stay at zero. I opened it in FlightPlot, and estimation looks normal.

Stifael, 2017-01-10:
It looks like the position controller is not active. while the attitude controller tracks the setpoints, the position controller is doing something rather strange (at least along the x and y axis).
[image]
Have you also tuned the vehicle for position control? I mean I think even with tuning it should do better than that.

weiweikong, 2017-01-10:
@Stifael With the same platform, we revert the ATT_EXT_HDG_M, and we could get excellent position control with the same PID parameters.

mhkabir, 2017-01-10:
Hmm. Odd. I have a hunch. 

Can you try the vision_pose mavros plugin to send the VICON pose instead of the mocap plugin? Make sure you set ATT_EXT_HDG_M to 1 for vision.

weiweikong, 2017-01-11:
@mhkabir As you suggest, we test it with VICON by using `vision_pose` plugin. The log is at
http://logs.uaventure.com/view/aMUNrWzfAmdEkg7qwNRSbC.

- `mocap/pose` --> `vision_pose/pose`,
- `LPE_VISION_ON = Enable`
- `ATT_EXT_HDG_M = 1`
- Fireware  1.5.1
- In Stabilize Mode, yaw angle drift.
- In POS CTRL Mode, turning circle.
Any suggestions?

mhkabir, 2017-01-11:
There is something fundamentally wrong with your setup, but I'm not sure what. Check your VICON system.
I just tested both the code paths, and they should be fine. You will need to debug this locally. I also suggest using latest master, because I don't know the state on 1.5.1

weiweikong, 2017-01-11:
When we set `ATT_EXT_HDG_M=1`, the VICON system and the quadrotor works well. So the problem is the heading information from VICON?

mhkabir, 2017-01-11:
Yes, seems so.

weiweikong, 2017-01-11:
@mhkabir Are you using the vicon_bridge from https://github.com/ethz-asl/vicon_bridge?

Stifael, 2017-01-27:
Whats the state here?

weiweikong, 2017-01-29:
@Stifael In our current experiments, this period still exist.

LorenzMeier, 2017-01-29:
@julianoes That should sound familiar.

julianoes, 2017-02-02:
@LorenzMeier I believe this is different because in my case the attitude rate setpoints were not published for a while and I don't see that here.

fboriero, 2017-02-07:
I think I have the same problem here! I followed all the steps that you described. I am using Optitrack and this is my result: 
[image]

LorenzMeier, 2017-04-10:
We need to work this one out.

LorenzMeier, 2017-04-10:
Is this issue still persisting?

weiweikong, 2017-04-10:
still exist, but not significantSent from Netease Mail
            On 04/10/2017 14:56, Lorenz Meier wrote: We need to work this one out.

fboriero, 2017-04-10:
Finally I decided to re-implement a position controller in python-dronekit so I left Px4. At this time I am in Caltech for a visiting period but In June I will be back in Italy and, If it still needed, I will able to help on this. 

Fabrizio

deksprime, 2017-04-19:
Sometimes, we also experience  problems when switching to offboard mode. In this log, when you look at local position x, y and z, you can see a random setpoint assignment for a very small time frame and then the setpoint comes back to original local position values, which it was supposed to directly apply when switched to offboard mode. Here is the log: 

http://logs.uaventure.com/view/LMZkMnjXTgVAT54A3gqhef

weiweikong, 2017-04-23:
@deksprime Correct. We set the rotors on the ground then switch to OFFBOARD mode,  then there is a sudden jump even though the OFFBOARD set local position is set to the same position.

LorenzMeier, 2017-05-01:
Still important, but I'm pushing this out. @weiweikong I don't have an off board setup here. Would you be able to debug this? Otherwise it will take some time to set up for this.

mzahana, 2017-12-22:
I have access to OptiTrac mocap and can try from my side. I will start with stable v1.7.

LorenzMeier, 2018-01-06:
@mzahana Could you update us on this? Thanks!

mzahana, 2018-01-06:
Things seem fine, at least for indoor test with mocap using EKF2.

Here is a log,
https://logs.px4.io/plot_app?log=60be36d4-ed01-434d-894a-7687eadfaced

LorenzMeier, 2018-01-06:
Given this proof point I'm closing this. @hamishwillee Could you please check that our documentation for MOCAP / Indoors points people to EKF2 now?

closed 2018-01-06 by LorenzMeier

mzahana, 2018-01-07:
For documentation purposes, here are the steps I followed.
* While defining the rigid body in the mocap software, I aligned forward the direction of the vehicle with the x-axis of the mocap system (OptiTrack)
* in the mocap software, Motive in my case, I set the software to send pose data with Z-axis up
* Used EKF2 estimator
* `EKF2_AID_MASK` set to enable fusion of vision pose and yaw, and to deactivate GPS (for mocap use)
* adjust primary height sensor to `vision`
* tune `EKF2_EV_DELAY` to reduce *ekf2_innovations* ( you can see that on the logs). Usually if feeding mocap data at high rate, reduce `EKF@_EV_DELAY` from the default value. I used a value of `50 ms` which worked quite well.
* adjust `EKF2_EV_POS_X`/Y/Z to represent the location of the center of the markers defining  rigid body, with respect to the vehicle frame, e.g. the FCU center.
* used `vrpn_ros_client` package to get pose data from mocap to ROS
```bash
roslaunch vrpn_client_ros sample.launch server:=<mocap_machine_ip>
```
* use `topic_tools` package to `relay` pose data to `/mavros/vision_pose/pose` topic
```bash
rosrun topic_tools relay /vrpn_client_node/<rigid_body_name>/pose /mavros/mocap/pose
```

Hope this helps!

fboriero, 2018-01-07:
Great!
Thanks a lot!

Fabrizio

eleboss, 2018-07-17:
Hi everyone, 
Have this problem been solved?
I currently using the latest pixhawk4 and V1.8.0 firmware to test offboard control mode, and troubled by this problem.
My drone will sudden jump to the air when I switch to OFFBOARD by using RC switch. 
I am using velocity control, but it seems the velocity control is not working until the jump procedure finished(reached the tko_speed) , this crush my drone at least 3 times. Because the default MPC_TKP_SPEED is 1.5m/s and the distance of the rooftop to the floor is only 3m, the drone reach to the top very fast then crash.  Then I trying to set the MPC_TKP_SPEED to 0 by force set, but the jump look no different with 1.5 m/s, crash again.
Finally, I found 1m/s, which is in the legal range of MPC_TKP_SPEED, works, the drone still jump, but not so high,.
I really want to find a good way to solve it, at least increase the legal range of MPC_TKP_SPEED, making it not jumping so high.