PX4/PX4-Autopilot #6115: Add support for JSON definition files of params, remove c files

LorenzMeier, 2016-12-21:
@simonegu @bkueng Parameters are not actually compiled, but a script picks them up from the C files, assembles the XML file and then generates code for the parameter storage. We just kept the files being .c files because that's what they originally were. We need to transition to a better format (e.g. JSON) and then move these over, because people think its C (but its not, the define does nothing).
https://github.com/PX4/Firmware/blob/97ce9edcac88e47fd406702ca92cf99dfa3a7e04/src/drivers/px4fmu/px4fmu_params.c

vooon, 2016-12-22:
I think a bit better to convert all that c-files to xml, because you already has description format used by qgc.
Only what you need is xml-merger-and-macro-processor to make single parameters.xml for selected build.
Some thing like xacro, where you pass `<xacro:include />` for all description files.

bkueng, 2018-02-21:
Still relevant

hamishwillee, 2018-11-12:
@dagar And relevant to our discussion on translation ...

dagar, 2018-11-13:
Notes
 - param metadata ultimately generates a single xml. One option to consider is storing the individual metadata pieces per module in xml, then simply merging.
 - we now have a module.yaml file that might be useful here (example https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/module.yaml) 
 - overall it's worth doing a quick comparison of xml, json, yaml, toml for this purpose

hamishwillee, 2018-11-13:
And while we're here, let's take the time to consider that we want to be able to translate the parameters, and ideally package them such that perhaps those translations could be fetched by QGC as a separate package from the firmware

bkueng, 2018-11-15:
> we now have a module.yaml file that might be useful here (example https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/module.yaml)

Yes module.yaml can already be used for that purpose, everything is already there to transition over existing modules.

closed 2019-02-27 by stale[bot]