PX4/PX4-Autopilot #6118: Dependencies for version generation broken

LorenzMeier, 2016-12-21:
@lucasdemarchi @bkueng The dependencies are currently broken on master:

```
make px4fmu-v1_default upload
```

Fails on a *clean* build with this error:

```
dir/version.c.obj.d -o src/lib/version/CMakeFiles/lib__version.dir/version.c.obj   -c ../src/lib/version/version.c
../src/lib/version/version.c: In function 'px4_firmware_version':
../src/lib/version/version.c:152:31: error: 'PX4_GIT_TAG_STR' undeclared (first use in this function)
  return version_tag_to_number(PX4_GIT_TAG_STR);
                               ^
compilation terminated due to -Wfatal-errors.
```

While doing:

```
make px4fmu-v1_default
make px4fmu-v1_default upload
```

works.

MaEtUgR, 2016-12-21:
On an other computer i didn't encounter that anymore, I will reopen if I see it again.

closed 2016-12-21 by MaEtUgR

MaEtUgR, 2016-12-22:
Double checked, with the newest version it's solved.

PR #6119 (2016-12-21): Fix version dependency
Should fix the error:
```
../src/lib/version/version.c:152:31: error: 'PX4_GIT_TAG_STR' undeclared (first use in this function)
  return version_tag_to_number(PX4_GIT_TAG_STR);
```

Fixes #6118