PX4/PX4-Autopilot #6120: Baro fail when starting sitl

MaEtUgR, 2016-12-21:
the red message `ERROR [sensors] Baro #0 fail:  STALE!` when starting sitl occurs apart from the changes in `/src/lib/ecl/validation/data_validator_group.cpp` from the following [commit](https://github.com/PX4/ecl/commit/6f78ef408425d5b9bc7e0eaceb1af9e5fd73b424).

As I see it this can come from an introduced bug in the commit OR a faulty simulator baro that doesn't really get validated by the new correct implementation...

LorenzMeier, 2016-12-21:
I think its because SITL can take longer to initialise. We need to somehow relax things there a bit.

bkueng, 2016-12-22:
It could also be because there is no noise on the simulated baro. Looking at `listener sensor_baro`, the values are constant before takeoff.

julianoes, 2016-12-24:
It also happens on landing, and in Gazebo and jMAVSim.

LorenzMeier, 2016-12-24:
The baro had no simulated noise. So the detection was correct. Fixed here:
https://github.com/PX4/Firmware/pull/6162

closed 2016-12-24 by LorenzMeier