PX4/PX4-Autopilot #6121: PX4 parameter generation generates parameters for the full code

mazahner, 2016-12-21:
Dear PX4 Team

The current Parameter generation scheme generates parameters (in the xml) for the full codebase in Firmware/src.

the relevant parameters are only filtered out when writing to px3_parameters.h/.c 

The parameter generation currently fails in the xml step, if two modules define the same parameter, even though one of them would not be in scope (in the .cmake file)
Is there any intent on making the xml generation more selective based on the scope? Would you accept such a PR?
Best, markus

LorenzMeier, 2016-12-21:
A PR for this would be fantastic. Certainly something we would like to do.

dagar, 2017-01-02:
@mazahner are you interested in adding support for this? We could certainly help point you in the right direction and test/review.

mazahner, 2017-01-03:
I think i have a solution working that works for me, but for testing in the various envionments it would be awesome to get some support. let me create a PR so you can give me some feedback.

mazahner, 2017-01-03:
PR sent

LorenzMeier, 2017-01-04:
Thanks! Fixed on master.

LorenzMeier, 2017-01-04:
@dagar @davids5 This also made the param generation step a lot faster. Will help with CI runs.

closed 2017-01-04 by LorenzMeier