PX4/PX4-Autopilot #6123: Arming denied due to GeoFence despite GPS Available

tops4u, 2016-12-21:
Today I wanted to run some test in the basement. So I got a lot of On and Off GPS Locks. Still at some points the GPS Reception was very acceptable. I wanted to Arm and shortly start the Copter to test the behaviour of the Home Position Altitude of the MinimOSD. But the Copter denied Arming with a Warning that a valid Home Position has to be available when using Geofencing RTL. But GPS Reception was ok, I got a valid Lock on my Home having 6 Sats available. Also Dis- / Rearming IO or Power Cycling did not help. 

So either GPS Quality did not reach some criteria that I don't know of? Or there is a Problem with the Arming when using Geofence RTL. As soon as I switch Geofence Violation to any other than RTL I can arm. 

Currently on 1.5.2 Stable. 

[image]

LorenzMeier, 2016-12-21:
Interesting experiment. The system correctly thinks that your GPS is potentially sketchy. Let us know if that test fails outdoors. But we're not going to change anything for indoor operation - its inherently unsafe and GPS is not designed to work there.

closed 2016-12-21 by LorenzMeier

tops4u, 2016-12-21:
I agree. Any hint about the criterias taht leads the System to think that GPS is unreliable. An older uBlox 6 might as well only reveal 6 Sats with the HDOP from the M8N above. So what is the Limit?

kd0aij, 2016-12-21:
Is this what you're looking for? https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L2083

tops4u, 2016-12-22:
Yes. I was looking around in this file and tried to figure out what makes a HomePoint valid. I figured it should be HDOP < 2.5 (AFAIK). In the OSD I could see that the Homepoint Alt was reset to 0 as Baro alone (which is active as long as GPS does not have a Lock) actually is about -310m off. 
I'll try to see how this behaves Outdoor, but for me it is still not clear why with this GPS Reception the Homepoint was not set valid.

kd0aij, 2016-12-22:
There are two parameters which specify the minimum gps quality:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L1210
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L1800

You'll have to check the values for your config to see what they are.

tops4u, 2016-12-22:
Those: 

static float eph_threshold = 5.0f;
static float epv_threshold = 10.0f;

I assume this corresponds to HDOP/VDOP?

kd0aij, 2016-12-22:
that's actually not clear to me from QGC; would have to look at the code.