PX4/PX4-Autopilot #6128: Build broken after NuttX merge

lucasdemarchi, 2016-12-22:
Build is currently broken for me when using the defaul Makefile generator instead of ninja. #6127 should help, making it to use ninja on Fedora, but it's not the proper fix.

More info:
https://gist.github.com/90bb7cc0fc49ab93f58aacf27c109dc8

It looks like it's spawning 3 instances of the rsync binary. When building with ninja it spawns only rsync only once, as it looks like to be the intention. However for make, I get several processes. Following command may help to reproduce the issue:

```
NO_NINJA_BUILD=1 strace -f -e execve make -j4 aerofc-v1_default 2>&1 | grep rsync
```

And this may be a bug in cmake, so here its version:

```
$ cmake --version
cmake version 3.6.2
```

mhkabir, 2016-12-22:
Non-Ninja build is broken for me as well. Ubuntu 14.04.
```
$ cmake --version
cmake version 3.2.2

```

matanhavi, 2016-12-23:
Same issue for me,
cmake version 3.4.1

dagar, 2016-12-27:
@lucasdemarchi should be resolved on master. Can you verify?

closed 2016-12-27 by LorenzMeier