PX4/PX4-Autopilot #6129: Companion Link broken for latest master

ecmnet, 2016-12-22:
After building the latest master, I am not able to connect my companion to the flight controller any more (PixRacer, 921600 8N1). Did I miss something here? I tried also 57800 baud, but could not achieve any connection.

mhkabir, 2016-12-22:
Did you verify that the SYS_COMPANION parameter is still correctly set?

ecmnet, 2016-12-22:
@mhkabir Of course I did: It is set properly to 921600 which is companion at 921600 baud.

ecmnet, 2016-12-22:
Switching back to 1.5.2 solves that issue. So I assume something regarding the px4fmu-V5 or the NUTTX changes is responsible for that.

LorenzMeier, 2016-12-22:
Thanks for letting us know. @davids5 Can you have a look at UART handling?

LorenzMeier, 2016-12-22:
Its the shell expansion:
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L623

`$MAVLINK_COMPANION_DEVICE` needs to become `${MAVLINK_COMPANION_DEVICE }`

LorenzMeier, 2016-12-23:
Fixed here: https://github.com/PX4/Firmware/pull/6148

closed 2016-12-23 by LorenzMeier

ecmnet, 2016-12-23:
@LorenzMeier @davids5  Just tested latest master:` Latest commit d00750c  `, but it seems as if TELEM2 still does not work.

LorenzMeier, 2016-12-23:
@ecmnet Thanks for testing! I think you might be hitting this one: https://github.com/PX4/Firmware/issues/6149

davids5, 2016-12-24:
@ecmnet - PR https://github.com/PX4/Firmware/pull/6160 should fix this issue.

ecmnet, 2016-12-26:
@LorenzMeier @davids5 I figured out, that with latest master the companion link works on PixRacer, if I use /dev/ttyS0 (the ESP8266 port) instead of TELEM2 at /dev/ttyS2. One obvious difference is that on ttyS0 rate multiplier is 1.0 but on /dev/ttyS2 it is 0.05. So far I can use /dev/ttyS0 as a workaround, although the connector is suboptimal.

LorenzMeier, 2016-12-26:
@ecmnet Thanks, that's extremely helpful, thank you so much for digging into it!

LorenzMeier, 2016-12-26:
@ecmnet Ok, so after digging into this a bit more: `CRTSCTS` is defined incorrectly in NuttX, so you effectively disabled flow control. I have a fix here:
https://github.com/PX4/Firmware/pull/6173

Let me know if that fixes it. You should either see flow control enabled or auto-disabled after a while.