PX4/PX4-Autopilot #6131: Land waypoint does not get executed

LorenzMeier, 2016-12-22:
@julianoes @AndreasAntener Our flight test team reports the land waypoint does not get executed. This is one of the logs:
http://review.px4.io/plot_app?log=caf379c8-5a27-48b7-b6e6-5a70656035d6

LorenzMeier, 2016-12-25:
[image]

@DonLakeFlyer I think this report was because of user error. The default value of 50 meters for the land waypoint is confusing users

closed 2016-12-25 by LorenzMeier

DonLakeFlyer, 2016-12-25:
What is Altitude supposed to mean on that command?

LorenzMeier, 2016-12-25:
The intended target altitude right now. It might require a compound command for multicopters with an approach waypoint and a land location waypoint. The target altitude is important because it tells the vehicle when to expect ground contact (e.g. go slow).

If you come down from 50 meters you don't want to come down with 1 m/s for the full 40 meters. But if you come down with 3 m/s on the ground you're going to hit hard. So what you want is to come down slow the last 10 meters.

This will be less of a concern when all vehicles have distance sensors, but right now they don't.

From a UX perspective you might want to combine two waypoints - the final location and the approach waypoint. And do the same for fixed wing.