PX4/PX4-Autopilot #6138: Parameter Profiles (Feature Request)

tops4u, 2016-12-22:
In my Opinion it would be nice to be able to have different Profiles which could be stored on the Flightcontroller for various Tasks (or in Ground Station).

Currently I have not found such a function in PX4 nor QGC. 

For example for a specific type of mission I would like to use a Ground LIDAR together with a Cruise speed of max 20kmh, smooth PID Values and a fixed YAW Heading in Mission Mode and Failsafe Action to RTL

In another type of mission I would like to set max Speed to 50kmh disregard the LIDAR, have YAW turn in direction to the next Waypoint and Failsafe Action to Land with a Geofence of 1000m and moderate PID Values due to different Payload.

Currently all the values have to be changed by Hand. Instead a Profile Switching (by MavLink Command or external Switch) would be desirable as it has i.e. been implemented in the BaseCam Gimbals where different Profiles can be stored dependent of the attached Camera.

DonLakeFlyer, 2016-12-22:
FYI: The latest version of QGC has a params section in a mission file. When you load a mission the params in the file are set as well. It's kind of an internal feature at the moment only used for GeoFence but in reality it will work for anything. If you move this issue into the QGC Issue we can continue talking there about making it part of the 3.1 release.

tops4u, 2016-12-22:
@DonLakeFlyer that would be nice. So the Mission Params would override the "Default Config" in the FC? I probably can't move this over. Shall I reopen in QGC?

DonLakeFlyer, 2016-12-22:
Open a new one in QGC

tops4u, 2016-12-22:
Continues in : https://github.com/mavlink/qgroundcontrol/issues/4333

tops4u, 2016-12-27:
As support for this is some kind limited in QGC I reopen this Case here.

I think as an Idea it would be nice to have a Default Set of parameters (as we have now), and then to have Profiles where deviations from the Defaults would overrule selected Parameters. For instance Tuining for specific Payload Setup, etc... 

Profiles should be selectable by RC Channel and/or bei QGC. 

Overriding Profiles would/should have a short Name and include a list of parameters that deviate from the Default Parameter Set. 

This would allow for rapid adaption to Mission Objectives and Payloads and minimize Errors in the Setup before Take off.

tops4u, 2016-12-27:
@LorenzMeier any votes on this one?

dagar, 2016-12-27:
You could have these as separate configs on the sdcard and select the variation you want by changing only SYS_AUTOSTART.

tops4u, 2016-12-27:
@dagar This looks like a possibility, allthough I'm not sure if it is a comfortable one. Any Hints (Docu?) how I could archive this? Is there a way to clone a Configuration to the SD?

dagar, 2016-12-28:
Some documentation - http://dev.px4.io/advanced-system-startup.html
The builtin airframes and corresponding init scripts are here - https://github.com/PX4/Firmware/tree/master/ROMFS/px4fmu_common/init.d

You could take the one that's close, duplicate it to the sdcard with a different number and select it by setting the SYS_AUTOSTART param.

tops4u, 2017-01-14:
@dagar I did not yet have time to test this... Is this a very uncommon case that Pilots might want to have different Profiles for different Use Cases? If so a more "friendly" Handling of this would be welcome instead of working with numbered Sys_Autostart Numbers?

dagar, 2017-01-14:
It really depends on the specifics. Some things can remain configured, but optionally enabled/disabled with a single param. If you have a significant number of param changes at what point should it be treated like an entirely new vehicle?

With the startup scripts you can easily have a base set of parameters, mixer, etc, and different configurations (SYS_AUTOSTART) that extend it.

Again depending on the specific situation it might make more sense for the PX4 module itself to support multiple configurations.

We could solve the problem at different levels. You can always just save/restore the entire param tree as well.

LorenzMeier, 2017-03-13:
This is very much a power user feature. You can already download and re-upload the complete param set via QGC and you can hand-edit the stored parameters in the QGC param files to create "diffs" of settings you just want to change.

Does that suffice as a workaround?

tops4u, 2017-03-13:
@LorenzMeier thanks for your answer. Yes it probably is. I was thinking if this would not be a requirement for professional (commercial) Drones as well as Payload and other Requirements change due to the nature of the Mission? 
I.e. I would have to disable a LIDAR when flying over Water, but fix YAW Heading when searching for fawn in Fields, or have different PID for various Payloads. If this is not a professional Requirement I can work with stored Params, this just adds some additional Housekeeping in order to keep them synced.
You may close this Issue if this is the case.