PX4/PX4-Autopilot #6140: LPE local/global position estimation drift

Stifael, 2016-12-22:
In SITL teseted:
After taking off in manual mode and disarming such that the vehicle remains on the floor, the LPE gives wrong local/global position estimates (keeps drifting)

jgoppert, 2016-12-22:
Sounds like landed fake measurement isn't kicking in, will take a look soon.

weiweikong, 2016-12-23:
@Stifael Do you mean there is a minimum take off speed if we switch the mode to OFFBOARD when the vehicle is on the ground?

In the position control code, there is a section manage it
```
/* special thrust setpoint generation for takeoff from ground */
	if (_pos_sp_triplet.current.valid
	    && _pos_sp_triplet.current.type == position_setpoint_s::SETPOINT_TYPE_TAKEOFF
	    && _control_mode.flag_armed) {

		// check if we are not already in air.
		// if yes then we don't need a jumped takeoff anymore
		if (!_takeoff_jumped && !_vehicle_land_detected.landed && fabsf(_takeoff_thrust_sp) < FLT_EPSILON) {
			_takeoff_jumped = true;
		}

		if (!_takeoff_jumped) {
			// ramp thrust setpoint up
			if (_vel(2) > -(_params.tko_speed / 2.0f)) {
				_takeoff_thrust_sp += 0.5f * dt;
				_vel_sp.zero();
				_vel_prev.zero();

			} else {
				// copter has reached our takeoff speed. split the thrust setpoint up
				// into an integral part and into a P part
				_thrust_int(2) = _takeoff_thrust_sp - _params.vel_p(2) * fabsf(_vel(2));
				_thrust_int(2) = -math::constrain(_thrust_int(2), _params.thr_min, _params.thr_max);
				_vel_sp_prev(2) = -_params.tko_speed;
				_takeoff_jumped = true;
				_reset_int_z = false;
			}
		}

		if (_takeoff_jumped) {
			_vel_sp(2) = -_params.tko_speed;
		}

	}
```
but I am not confirmed that this section results the drifiting.

Stifael, 2016-12-23:
@weiweikong: nope, it has nothing to do with offboard mode. The values that you can see from qgc (global and local position lat, long, and alt) are sent from the estimator and is independent of the position controller. Those values are changing even when on ground after landing and disarming. this behaviour does not occur when using ekf2 estimator

jgoppert, 2016-12-23:
I'm kind of being stretched thin. I'm trying to get the IEKF running as a replacement to LPE/ attitude_estimator_q and also keep maintaining LPE during the transition. @Stifael before I start debugging   can  you give me more info on if you were flying with optical flow or gps?

jgoppert, 2016-12-23:
@Stifael I'm unable to reproduce with GPS, will try flow.

jgoppert, 2016-12-23:
@Stifael 

```bash
make posix_sitl_default iris_opt_flow
```

takeoff and land using optical flow in SITl: http://logs.px4.io/plot_app?log=016bf446-ff28-4cc6-a287-695b9e33b42b

```bash
make posix_sitl_default iris_opt_flow
```
I just now saw it after running the optical flow sim:

```bash
pxh> commandINFO  [local_position_estimator] [lpe] baro init 0 m std -2147483648 cm
INFO  [commander] home: 47.3977432, 8.5455942, 0.10
INFO  [tone_alarm] home_set
er takeoff
pxh> INFO  [commander] Taking off
INFO  [tone_alarm] positive
INFO  [commander] home: 47.3977432, 8.5455942, 0.13
INFO  [tone_alarm] arming
INFO  [commander] Takeoff detected
INFO  [local_position_estimator] [lpe] land timeout 
INFO  [tone_alarm] neutral
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/iris/imu, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/iris/motor_speed/1, deleting message. This warning is printed only once.
WARN  [navigator] global position timeout
INFO  [tone_alarm] gps_warning
INFO  [commander] no gps
WARN  [commander] failsafe mode on
ERROR [sensors] Baro #0 fail:  STALE!
```

http://logs.px4.io/plot_app?log=d5496997-6bdc-4eb1-8c8d-84d9b80c3d2e

weiweikong, 2016-12-23:
@Stifael Thank you for your explain. I will read the code and test more to find the strange period, taking off aggressively, when uses LPE, as mentioned in issue https://github.com/PX4/Firmware/issues/6106.

PR #6146 (2016-12-23): LPE fault relaxation and sitl fix
Closes #6140.

@Stifael please test.

I was able to reproduce drift when the fuse gps flag was set to off. This occurred since the standard iris config for LPE didn't set the param and if you run the  "make posix_sitl_default gazebo_iris_opt_flow" followed by "make posix_sitl_default gazebo_iris" Then the iris without flow and with gps will have flow enabled and gps disabled. It will still take off due to the land detector correction, but then drift since it doesn't have a valid position measurement anymore.
closed 2016-12-23 by jgoppert