PX4/PX4-Autopilot #6145: Sensors Issue in Master on Pixracer

pkocmoud, 2016-12-22:
This has occurred on 2 new Pixracer boards. On Master the both IMUs appear to be starting, but the messages indicate 1 IMU cannot be found. Below is from the same board flashed with either Stable or Master.

** (GOOD) Stable Boot Messages:**
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
nsh: mount: mount failed: No such device
nsh: mkfatfs: mkfatfs failed: No such device
WARN  [param] selected parameter default file /fs/mtd_params
WARN  [rgbled] no RGB led on bus #1
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
INFO  [ver] match: PX4FMU_V4
INFO  [ver] match: PX4FMU_V4
WARN  [dataman] Could not open data manager file /fs/microsd/dataman
ERROR [dataman] dataman start failed
INFO  [ver] match: PX4FMU_V4
WARN  [drivers__boards__px4fmu-v4] reset done, 50 ms
ERROR [ms5611] driver start failed
MS5611_SPI on SPI bus 2 at 3 (20000 KHz)
WARN  [bst] no devices found
INFO  [ver] match: PX4FMU_V4
WARN  [hmc5883] no device on bus 2
read_reg failWARN  [lis3mdl] no device on bus 2
LIS3MDL_SPI on SPI bus 1 at 7 (11000 KHz)
LIS3MDL bad ID: ffWARN  [lis3mdl] no device on bus 3
HMC5883_SPI on SPI bus 1 at 5 (11000 KHz)
MPU6000 on SPI bus 1 at 6 (1000 KHz)
MPU9250 on SPI bus 1 at 4 (1000 KHz)
ERROR [bmi160] driver start failed
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
SF1XX disabled.
ERROR [sf1xx] driver start failed
WARN  [hmc5883] mag cal status changed
WARN  [commander] No RGB LED found at /dev/rgbled0
INFO  [ver] match: PX4FMU_V4
INFO  [mavlink] mode: Config, data rate: 20000 B/s on /dev/ttyS0 @ 921600B
WARN  [mavlink] hardware flow control not supported
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
INFO  [ver] match: PX4FMU_V4
INFO  [ver] match: PX4FMU_V4
px4flow [199:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
WARN  [sdlog2] [blackbox] no microSD card, disabling logging
WARN  [mavlink] Failed to open MAVLink log: /fs/microsd/msgs_2000_01_01_00_09_10.txt
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [lib__ecl] EKF IMU buffer length = 21
INFO  [lib__ecl] EKF observation buffer length = 16

NuttShell (NSH)
nsh> INFO  [lib__ecl] EKF using pressure height - commencing alignment
WARN  [mavlink] Disabling hardware flow control
WARN  [mavlink] Disabling hardware flow control
WARN  [commander_tests] PREFLIGHT FAIL: MAG #1 UNCALIBRATED
WARN  [commander_tests] PREFLIGHT FAIL: ACCEL #1 UNCALIBRATED
WARN  [commander_tests] PREFLIGHT FAIL: GYRO #1 UNCALIBRATED
WARN  [commander_tests] PREFLIGHT FAIL: GPS RECEIVER MISSING
INFO  [lib__ecl] EKF alignment complete

** (GOOD) On Stable:**
nsh> tests sensors
Running sensors tests:

  sensor: accel0
        ACCEL: test start
        ACCEL accel: x: -0.4505 y:  0.1026      z:  9.6928 m/s^2
        OK: ACCEL passed all tests successfully
  sensor: accel1
        ACCEL: test start
        ACCEL accel: x: -0.3393 y:  0.2597      z:  9.6162 m/s^2
        OK: ACCEL passed all tests successfully
  sensor: gyro0
        GYRO: test start
        GYRO rates: x:  0.0016  y:  0.0026      z:  0.0011 rad/s
        OK: GYRO passed all tests successfully
  sensor: gyro1
        GYRO: test start
        GYRO rates: x: -0.0080  y: -0.0043      z: -0.0175 rad/s
        OK: GYRO passed all tests successfully
  sensor: mag0
        MAG: test start
        MAG values: x:  0.0380  y: -0.1430      z: -0.5235
        OK: MAG passed all tests successfully
  sensor: baro0
        BARO: test start
        BARO pressure: 1004.8500 mbar   alt:  70.1586 m temp:  37.3500 deg C
        OK: BARO passed all tests successfully

** (BAD) Master Boot Messages:**
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
nsh: mount: mount failed: No such device
nsh: mkfatfs: mkfatfs failed: No such device
WARN  [param] selected parameter default file /fs/mtd_params
WARN  [rgbled] no RGB led on bus #1
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
INFO  [ver] match: PX4FMU_V4
INFO  [ver] match: PX4FMU_V4
WARN  [dataman] Could not open data manager file /fs/microsd/dataman
ERROR [dataman] dataman start failed
INFO  [ver] match: PX4FMU_V4
WARN  [drivers__boards__px4fmu-v4] reset done, 50 ms
MS5611_SPI on SPI bus 2 at 3 (20000 KHz)
WARN  [bst] no devices found
INFO  [ver] match: PX4FMU_V4
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [lis3mdl] no device on bus 2
LIS3MDL_SPI on SPI bus 1 at 7 (11000 KHz)
WARN  [lis3mdl] no device on bus 3
HMC5883_SPI on SPI bus 1 at 5 (11000 KHz)
MPU6000 on SPI bus 1 at 6 (1000 KHz)
MPU9250 on SPI bus 1 at 4 (1000 KHz)
ERROR [bmi160] driver start failed
INFO  [ver] match: PX4FMU_V4
MPU6000 on SPI bus 1 at 6 (1000 KHz)
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [lis3mdl] no device on bus 2
LIS3MDL_SPI on SPI bus 1 at 7 (11000 KHz)
WARN  [lis3mdl] no device on bus 3
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
WARN  [commander] No RGB LED found at /dev/rgbled0
INFO  [ver] match: PX4FMU_V4
INFO  [mavlink] mode: Config, data rate: 20000 B/s on /dev/ttyS0 @ 921600B
WARN  [mavlink] hardware flow control not supported
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
INFO  [ver] match: PX4FMU_V4
INFO  [ver] match: PX4FMU_V4
px4flow [207:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
WARN  [sdlog2] [blackbox] no microSD card, disabling logging
WARN  [mavlink] Failed to open MAVLink log: /fs/microsd/msgs_2000_01_01_00_11_29.txt
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [lib__ecl] EKF IMU buffer length = 21
INFO  [lib__ecl] EKF observation buffer length = 16

NuttShell (NSH)
nsh> WARN  [mavlink] Disabling hardware flow control
INFO  [lib__ecl] EKF using pressure height - commencing alignment
WARN  [mavlink] Disabling hardware flow control
WARN  [commander_tests] Not ready to fly: Sensors not set up correctly
WARN  [commander_tests] PREFLIGHT FAIL: MAG #1 UNCALIBRATED
WARN  [commander_tests] PREFLIGHT FAIL: ACCEL #0 UNCALIBRATED
WARN  [commander_tests] Warning: Primary accelerometer not found
WARN  [commander_tests] PREFLIGHT FAIL: GYRO #0 UNCALIBRATED
WARN  [commander_tests] Warning: Primary gyro not found
INFO  [lib__ecl] EKF alignment complete

** (BAD) On Master:**
tests sensors
Running sensors tests:

  sensor: accel0
        ACCEL: test start
        ACCEL accel: x:  0.0287 y:  0.4321      z:  9.6701 m/s^2
        OK: ACCEL passed all tests successfully
  sensor: accel1
        ACCEL: test start
        ACCEL: open fail, run <mpu6000 start> or <lsm303 start> or <bma180 start> first.
  sensor: gyro0
        GYRO: test start
        GYRO rates: x: -0.0104  y: -0.0015      z: -0.0168 rad/s
        OK: GYRO passed all tests successfully
  sensor: gyro1
        GYRO: test start
        GYRO: open fail, run <l3gd20 start> or <mpu6000 start> first.
  sensor: mag0
        MAG: test start
        MAG values: x:  0.0499  y: -0.1609      z: -0.5221
        OK: MAG passed all tests successfully
  sensor: baro0
        BARO: test start
        BARO pressure: 1004.9600 mbar   alt:  69.2368 m temp:  34.0700 deg C
        OK: BARO passed all tests successfully

pkocmoud, 2016-12-22:
Performing a calibration and/or rebooting does not resolves this issue.

pkocmoud, 2016-12-23:
nsh> mpu6000 -T 20608 start
MPU6000 on SPI bus 1 at 6 (1000 KHz)
nsh> mpu6000 stop
ERROR [mpu6000] bus 0 not started

davids5, 2016-12-23:
@pkocmoud  - were these tests done with SD cards installed?

pkocmoud, 2016-12-23:
They were not.

kd0aij, 2016-12-23:
https://github.com/PX4/Firmware/pull/6155#issuecomment-269009863

LorenzMeier, 2016-12-23:
I'm closing this one as the fix is in. Please let us know if this persists.

closed 2016-12-23 by LorenzMeier

pkocmoud, 2016-12-23:
Yes, I have had 2 very nice flights on Master just now. Thank you!