PX4/PX4-Autopilot #6149: NuttX 7.18 UART settings and/or RXDMA handling review

LorenzMeier, 2016-12-23:
@davids5 There seem to be issues with UART handling in general in multiple instances, most recently this one:
https://github.com/PX4/Firmware/pull/6147#issuecomment-268957737

@tumbili Would you mind printing all bytes you receive in HEX and doing the same with your driver when it works? It would be super helpful to see how the byte pattern gets corrupted.

tumbili, 2016-12-23:
@LorenzMeier Yeah, will do now.

tumbili, 2016-12-23:
@LorenzMeier @davids5 This is the radar output on latest master:
```
bytes: 48 a4 48 range: 50.00
bytes: 48 a4 80 range: 1.62
bytes: 48 80 48 range: 50.00
bytes: 48 a4 48 range: 50.00
bytes: 48 a4 80 range: 1.62
bytes: 48 80 48 range: 50.00
bytes: 48 a4 80 range: 1.62
bytes: 48 a4 48 range: 50.00
bytes: 48 a4 80 range: 1.62
bytes: 48 80 48 range: 50.00
bytes: 48 a4 48 range: 50.00
bytes: 48 80 48 range: 50.00
bytes: 48 a4 48 range: 50.00
bytes: 48 a4 80 range: 1.62
```

The first byte (48) indicated that the next two bytes will be the range data.
The last number is the measured range and the correct one is 1.62m, so 50.0 is the corrupt one.

Data from working version to come.

LorenzMeier, 2016-12-23:
Are you printing everything that's on the port or just what is coming through the if statement?

LorenzMeier, 2016-12-23:
I mean this if statement:
```
if ((buf[0] & 0x80) == 0x00 && (buf[0] & 0x7F) == RADAR_RANGE_DATA && len >= 3) {
```

I need the raw data without it.

tumbili, 2016-12-23:
@LorenzMeier Sorry, will come.

tumbili, 2016-12-23:
@LorenzMeier This is the buffer on pre-NuttX update (I'm printing the raw buffer)
```
48 a4 80 
48 a4 80 48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 48 a4 80 
48 a4 80 
48 a4 80 48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 
48 a4 80 

```

tumbili, 2016-12-23:
@LorenzMeier And this is the corrupted one:
```
a4 80 48 
a4 80 48 a4 80 48 
48 80 48 a4 80 48 
a4 80 48 
a4 80 48 
a4 80 48 
48 a4 48 
a4 80 48 
a4 80 48 
a4 80 48 
a4 80 48 48 a4 80 
a4 80 48 
a4 80 48 
a4 80 48 
a4 80 48 
a4 80 48 
48 80 48 
a4 80 48 
a4 80 48 
a4 80 48 
a4 80 48 
48 a4 48 a4 80 48 
a4 80 48 
a4 80 48 
a4 80 48
```

tumbili, 2016-12-23:
@LorenzMeier Ok, it seems you were right.

LorenzMeier, 2016-12-23:
Sort of halfway. Its BCA ordering plus occasional drops. @davids5 That's a red flag requiring inspection.

tumbili, 2016-12-23:
@LorenzMeier Anyway, I'll take this as a chance to increase my knowledge and will try to write a parser.
I think the sx0f driver does something similar, hope it inspires me.

LorenzMeier, 2016-12-23:
@davids5 Here is the diff between what we had in NuttX 6 and the NuttX 7 driver:

```diff
diff --git a/arch/arm/src/stm32/stm32_serial.c b/arch/arm/src/stm32/stm32_serial.c
index 644c810817..c4c272328a 100644
--- a/arch/arm/src/stm32/stm32_serial.c
+++ b/arch/arm/src/stm32/stm32_serial.c
@@ -1,7 +1,7 @@
 /****************************************************************************
  * arch/arm/src/stm32/stm32_serial.c
  *
- *   Copyright (C) 2009-2014, 2016 Gregory Nutt. All rights reserved.
+ *   Copyright (C) 2009-2013 Gregory Nutt. All rights reserved.
  *   Author: Gregory Nutt <gnutt@nuttx.org>
  *
  * Redistribution and use in source and binary forms, with or without
@@ -63,12 +63,12 @@
 #include "chip.h"
 #include "stm32_uart.h"
 #include "stm32_dma.h"
-#include "stm32_rcc.h"
 #include "up_arch.h"
 #include "up_internal.h"
+#include "os_internal.h"
 
 /****************************************************************************
- * Pre-processor Definitions
+ * Definitions
  ****************************************************************************/
 /* Some sanity checks *******************************************************/
 /* DMA configuration */
@@ -77,7 +77,7 @@
  * have also been selected.
  */
 
-#ifdef SERIAL_HAVE_DMA
+#if SERIAL_HAVE_DMA
 
 #  if defined(CONFIG_STM32_STM32F20XX) || defined(CONFIG_STM32_STM32F40XX)
 /* Verify that DMA has been enabled and the DMA channel has been defined.
@@ -150,7 +150,7 @@
 #    endif
 
 #  elif defined(CONFIG_STM32_STM32L15XX) || defined(CONFIG_STM32_STM32F10XX) || \
-        defined(CONFIG_STM32_STM32F30XX) || defined(CONFIG_STM32_STM32F30XX)
+        defined(CONFIG_STM32_STM32F30XX)
 
 #    if defined(CONFIG_USART1_RXDMA) || defined(CONFIG_USART2_RXDMA) || \
       defined(CONFIG_USART3_RXDMA)
@@ -159,9 +159,9 @@
 #      endif
 #    endif
 
-#    if defined(CONFIG_UART4_RXDMA) || defined(CONFIG_UART5_RXDMA)
+#    if defined(CONFIG_UART4_RXDMA)
 #      ifndef CONFIG_STM32_DMA2
-#        error STM32 UART4/5 receive DMA requires CONFIG_STM32_DMA2
+#        error STM32 USART4 receive DMA requires CONFIG_STM32_DMA2
 #      endif
 #    endif
 
@@ -170,24 +170,28 @@
 #    define DMAMAP_USART1_RX  DMACHAN_USART1_RX
 #    define DMAMAP_USART2_RX  DMACHAN_USART2_RX
 #    define DMAMAP_USART3_RX  DMACHAN_USART3_RX
-#    define DMAMAP_UART4_RX   DMACHAN_UART4_RX
-#    define DMAMAP_UART5_RX   DMACHAN_UART5_RX
+#    define DMAMAP_UART4_RX   DMACHAN_USART4_RX
 
 #  endif
[... 1821 lines cut ...]
+ *
+ ****************************************************************************/
+static void up_onrxdeque(struct uart_dev_s *dev)
+{
+  struct up_dev_s *priv = (struct up_dev_s*)dev->priv;
+
+  if (priv->iflow && (priv->rts_gpio != 0))
+    {
+      // We've delivered the chars to the OS FIFO now update RTS state as needed
+      ssize_t numUsed = uart_numrxavail(dev);
+      uint8_t threshold = 32;
+      if (threshold > priv->dev.recv.size/2) {
+	threshold = priv->dev.recv.size/2;
+      }
+      // If we have 32 bytes free in the buffer then we tell the
+      // sender they can resume
+      bool wantResume = (priv->dev.recv.size - numUsed >= threshold);
+
+      if(wantResume)
+        stm32_gpiowrite(priv->rts_gpio, false); // nRTS low means ready to recv
+    }
 }
 #endif

```

davids5, 2016-12-24:
@LorenzMeier - PR https://github.com/PX4/Firmware/pull/6160 is in and tested.

LorenzMeier, 2016-12-24:
@tumbili Can you re-test? This should be fixed on master now.

ecmnet, 2016-12-24:
@LorenzMeier Just again did a test with latest master and still no TELEM2 connection to the companion. Anything to adjust regarding params?

LorenzMeier, 2016-12-24:
@ecmnet Try a `make distclean` (but be aware that this wipes any unstaged files in the directory, so make sure to commit everything before). It seems to work for me:

```
instance #0:
        GCS heartbeat:  328369 us ago
        mavlink chan: #0
        type:           3DR RADIO
        rssi:           218
        remote rssi:    219
        txbuf:          91
        noise:          61
        remote noise:   56
        rx errors:      0
        fixed:          0
        flow control:   ON
        rates:
        tx: 0.549 kB/s
        txerr: 0.000 kB/s
        rx: 0.021 kB/s
        rate mult: 0.381
        accepting commands: YES
        MAVLink version: 2
        transport protocol: serial (/dev/ttyS1 @57600)

instance #1:
        mavlink chan: #1
        no telem status.
        flow control:   OFF
        rates:
        tx: 0.000 kB/s
        txerr: 0.136 kB/s
        rx: 0.060 kB/s
        rate mult: 0.050
        accepting commands: YES
        MAVLink version: 1
        transport protocol: serial (/dev/ttyS2 @57600)

instance #2:
        GCS heartbeat:  174801 us ago
        mavlink chan: #2
        type:           USB CDC
        flow control:   OFF
        rates:
        tx: 15.191 kB/s
        txerr: 0.000 kB/s
        rx: 0.017 kB/s
        rate mult: 1.000
        accepting commands: YES
        MAVLink version: 1
        transport protocol: serial (/dev/ttyACM0 @2000000)
```

ecmnet, 2016-12-24:
@LorenzMeier I always do. Mavlink status seems to be fine, although I get no connection:

[image]

LorenzMeier, 2017-01-04:
This is all fixed since a few days on master. Thanks everybody for helping debugging this!

closed 2017-01-04 by LorenzMeier