PX4/PX4-Autopilot #6152: PREFLIGHT FAIL: AIRSPEED SENSOR MISSING repeatedly (no airframe set)

DonLakeFlyer, 2016-12-23:
* Flashed latest master
* Airframe is not set yet

Get continuous stream of: PREFLIGHT FAIL: AIRSPEED SENSOR MISSING

DonLakeFlyer, 2016-12-23:
The speed at which these PREFLIGHT FAIL message repeate is pretty annoying. I set the airframe and now I get GYRO 1 UNCALIBRATED about 1 a second or so. Have to keep closing the message in QGC over and over and over and over again!

LorenzMeier, 2016-12-23:
I don't have that on a new board. Can you run "param reset" and see what happens?

DonLakeFlyer, 2016-12-23:
Same thing after param reset.

LorenzMeier, 2016-12-23:
What kind of board is that?

LorenzMeier, 2016-12-23:
And what links did you connect? That also plays a role.

DonLakeFlyer, 2016-12-23:
3DR Pixhawk connected over USB

DonLakeFlyer, 2016-12-23:
FYI: I just flashed ArduPilot and then flashed back to PX4 and although airframe was not set I did not get it this time. Tried a param reset and it didn't come back. So not quite sure what kicks it off.

dagar, 2017-04-11:
Have you seen this recently? I'm not sure what the underlying problem is.

closed 2018-02-21 by PX4BuildBot