PX4/PX4-Autopilot #6153: START_RX_PAIR and REQUEST_AUTOPILOT_CAPABILITIES do not send COMMAND_ACK

DonLakeFlyer, 2016-12-23:
This causes QGC to go into its command retry logic since it thinks the command was lost on the wire. There may be other COMMAND_LONGs with this problem as well that I haven't hit with QGC yet.

DonLakeFlyer, 2016-12-23:
Also just hit the problem with MAV_CMD_PREFLIGHT_CALIBRATION

DonLakeFlyer, 2016-12-23:
Two options:
1) If someone can do a code inpsection of firmware and give me the list of command which have this problem I can add a hack to QGC to not run retry logic for these commands.
2) If #1 is not possible I'm going to need to turn off retry logic for all commands when running PX4 firmware. Kind of a big loss from a hardening over lossy connection standpoint.

mhkabir, 2016-12-23:
I think we should just make sure we properly ACK all commands.

DonLakeFlyer, 2016-12-23:
I agree but QGC needs to support firmware version which will not have the fix. 

FYI: I already implemented #2 in QGC since at this point I had no other choice and things were fairly broken without it.

LorenzMeier, 2016-12-23:
@DonLakeFlyer Can you revert QGC and test this instead?
https://github.com/PX4/Firmware/pull/6154

Either revert QGC completely or if you want to support old FW versions do a version check. Don't keep things broken just to support old Firmware builds - that's not our strategy for PX4.

Also please fix QGC to ask for an ACK. The commands QGC sends are not asking for one yet.

DonLakeFlyer, 2016-12-23:
> Also please fix QGC to ask for an ACK. The commands QGC sends are not asking for one yet.

What is the deal with COMMAND_LONG.confirmation? Docs say:
```0: First transmission of this command. 1-255: Confirmation transmissions (e.g. for kill command)```

Which says nothing about it controlling whether you want an Ack back or not. According to current doc setting confirmation=0 would seem like the right thing.

If you want to force people to latest master and next release then I can revert QGC otherwise I'll need version check. Up to you.

DonLakeFlyer, 2016-12-23:
Also FYI: Not running retry logic doesn't really break things. It just is less user friendly to lossy connections. Up until a few weeks ago that was the way QGC always ran.

LorenzMeier, 2016-12-24:
This is fixed.

closed 2016-12-24 by LorenzMeier