PX4/PX4-Autopilot #6158: mavros WP reached spamming in SITL

jgoppert, 2016-12-23:
I think we are all tired of seeing this:

```bash
[ INFO] [1482530619.053705307, 98.624000000]: WP: reached #1
[ INFO] [1482530619.155152754, 98.726000000]: WP: reached #1
[ INFO] [1482530619.256826161, 98.828000000]: WP: reached #1
[ INFO] [1482530619.358218357, 98.928000000]: WP: reached #1
[ INFO] [1482530619.459594025, 99.030000000]: WP: reached #1
[ INFO] [1482530619.560925117, 99.132000000]: WP: reached #1
[ INFO] [1482530619.662820135, 99.232000000]: WP: reached #1
[ INFO] [1482530619.764426380, 99.334000000]: WP: reached #1
[ INFO] [1482530619.865761188, 99.436000000]: WP: reached #1
[ INFO] [1482530619.967476256, 99.538000000]: WP: reached #1
[ INFO] [1482530620.069147175, 99.638000000]: WP: reached #1
[ INFO] [1482530620.170544629, 99.740000000]: WP: reached #1
[ INFO] [1482530620.272065041, 99.842000000]: WP: reached #1
```

How do we get rid of it. I put a print statement when commander sends a waypoint reached message, and mavros handles each waypoint reached message by printing the above message. So why is mavros printing so many messages.

LorenzMeier, 2016-12-23:
@julianoes @bkueng The system load also rises in mission mode by 10%, which makes no sense from a control point of view. I think the recent changes for some of the external APIs led to some topics updating beyond the necessary.

dagar, 2016-12-23:
This is coming from the mavros waypoint plugin and MISSION_ITEM_REACHED - https://github.com/mavlink/mavros/blob/cafb05e79697fc8f1cb889ffac9e5b9d68183460/mavros/src/plugins/waypoint.cpp#L366

julianoes, 2016-12-24:
@dagar you're right. I think `MISSION_ITEM_REACHED` should only be sent once on reached because we already send `CURRENT_MISSION_ITEM` every second.

LorenzMeier, 2016-12-25:
@julianoes You introduced this here:
https://github.com/PX4/Firmware/blob/92c35f26be05d61f557e6b092359efaa42e2faac/src/modules/mavlink/mavlink_mission.cpp#L378-L380

I recall that you were trying to solve the issue of this very packet dropping and your external mission state machine not keeping track. Can you go back to your code and see if the current waypoint is sufficient and we can remove this?

julianoes, 2016-12-25:
@LorenzMeier right, this was back in the days when QGC showed the current waypoint and allowed to click on a waypoint in the list to set the current. Since this functionality is all gone, it's probably not needed anymore.

LorenzMeier, 2016-12-25:
Functionality is still there. In the flight view.

julianoes, 2016-12-25:
I guess in that case mavros should get rid of the printf.

LorenzMeier, 2016-12-25:
Fixed here: https://github.com/PX4/Firmware/pull/6168/commits/6e323047fb6ba0834ca4b4fc8f02827d12217d56

closed 2016-12-25 by LorenzMeier