PX4/PX4-Autopilot #6159: INT32 param cast failing?

pickledgator, 2016-12-24:
Having some funky problems with int32 parameters that are cast from mavlink to a param getting corrupted. 

[Specifically, param.c](https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param.c#L558-L561):
`case PARAM_TYPE_INT32:
	params_changed = params_changed || s->val.i != *(int32_t *)val;
        printf("in param.c %f %d %d\n", *(float *)val, *(int32_t *)val);
	s->val.i = *(int32_t *)val;
	break;
`

The float parameters seem to cast fine, but *(int32_t *)val is not converting float values to ints properly. I'm on OSX 10.12.2, maybe a clang update broke this?

Console print from above yields:
>> in param.c 1.000000 1065353216

LorenzMeier, 2016-12-24:
Are you aware that the MAVLink spec calls for a binary transfer of the param (not casted value, which looses precision)? What software component is sending the param on the other end?

http://qgroundcontrol.org/mavlink/parameter_protocol

pickledgator, 2016-12-24:
Thanks Lorenz, using the param union solved it. That was the part that I was missing on the send side.

closed 2016-12-24 by pickledgator