PX4/PX4-Autopilot #6163: mavlink stop-all not working.

davids5, 2016-12-24:
master @ https://github.com/PX4/Firmware/commit/5607df8a4434ca3ca8b01f30efe1efe53473649f
PX4fmu v2 - 2.4.5 (ish) HW
Only USB and DEBUG cables connected. 
SD card in/out does not matter. 
mtd erased

SD card out
```
[boot] Fault Log info File No 3 Length 3257 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
nsh: mount: mount failed: No such device
nsh: mkfatfs: mkfatfs failed: No such device
WARN  [param] selected parameter default file /fs/mtd_params
WARN  [param] error importing from '/fs/mtd_params'
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
  BAT_N_CELLS: curr: 0 -> new: 3
WARN  [dataman] Could not open data manager file /fs/microsd/dataman
ERROR [dataman] dataman start failed
MS5611_SPI on SPI bus 4 at 2 (20000 KHz)
WARN  [ms5611] no device on bus 4
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst] no devices found
INFO  [ver] match: PX4FMU_V2
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [lis3mdl] no device on bus 2
WARN  [hmc5883] no device on bus 2 (type: 1)
WARN  [mpu6000] no device on bus 3
WARN  [mpu6000] no device on bus 5
WARN  [mpu6000] no device on bus 5
WARN  [mpu9250] no device on bus 4
MPU6000 on SPI bus 1 at 4 (1000 KHz)
WARN  [mpu6000] no device on bus 5
WARN  [mpu9250] no device on bus 3
WARN  [mpu9250] no device on bus 4
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
INFO  [px4io] default PWM output device
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
INFO  [ver] match: PX4FMU_V2
px4flow [127:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
WARN  [sdlog2] [blackbox] no microSD card, disabling logging
WARN  [mavlink] Failed to open MAVLink log: /fs/microsd/msgs_2000_01_03_05_42_10.txt
No autostart ID found

NuttShell (NSH)
nsh> WARN  [mavlink] Disabling hardware flow control
WARN  [mavlink] Disabling hardware flow control
WARN  [commander] Not ready to fly: Sensors not set up correctly

nsh> mavlink stop-all
WARN  [mavlink] waiting for instances to stop
.........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................ERROR [mavlink] Couldn't stop all mavlink instances.
nsh>
```

LorenzMeier, 2016-12-26:
Works for me - was this maybe the result of the previous flow control issue? Can you try to repro on master and close if fixed?

```
nsh> mavlink stop-all
WARN  [mavlink] waiting for instances to stop
................WARN  [mavlink] exiting channel 0
........WARN  [mavlink] exiting channel 1
...................WARN  [mavlink] exiting channel 2

WARN  [mavlink] all instances stopped
nsh> 
```

western-co-at, 2017-01-03:
Some time ago I had a similar problem with stopping all mavlink interfaces.
The reason was, the the USB's mavlink exit at those lines during the initialisation:
https://github.com/PX4/Firmware/blob/000d965a5ee1db79117bcd88894d85dfec6f8b86/src/modules/mavlink/mavlink_main.cpp#L1893-L1896
and so didn't bumb the _mavlink_instances count. When running the top command I could see running two mavlink_if1 processes but only one mavlink_rcv_if1. After a mavlink stop-all I had still running one single mavlink_if1 process which I couldn't terminate by a stop-all.

davids5, 2017-01-03:
Working on current master.

```
 ver all
HW arch: PX4FMU_V2
FW git-hash: 000d965a5ee1db79117bcd88894d85dfec6f8b86
FW version: 1.5.2 0 (17105408)
OS: NuttX
OS version: Release 1.8.0 (17301759)
Build datetime: Jan  3 2017 09:19:37
Build uri: localhost
Toolchain: GNU GCC, 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
MCU: STM32F42x, rev. 3
UID: 400038:33355110:32383336
```

closed 2017-01-03 by davids5