PX4/PX4-Autopilot #6165: ALTITUDE.altitude_relative value incorrect

DonLakeFlyer, 2016-12-24:
Running SITL on a master from yesterday. You can see the problem in the mavlink inspector screenshot. Look at the relative alt from the GPS which is correcvt, as opposed to relative alt in ALTITUDE which is incorrect:
[image]

LorenzMeier, 2016-12-25:
I can't reproduce this with current master, with none of the estimators in no state (home set or home not yet set):

[image]
[image]
[image]
[image]

LorenzMeier, 2016-12-25:
Diff to debug locally:

```C++
diff --git a/src/modules/mavlink/mavlink_messages.cpp b/src/modules/mavlink/mavlink_messages.cpp
index f525f6f613..9a9570996a 100644
--- a/src/modules/mavlink/mavlink_messages.cpp
+++ b/src/modules/mavlink/mavlink_messages.cpp
@@ -3429,6 +3429,7 @@ protected:
                        } else {
                                msg.altitude_relative = NAN;
                        }
+                       PX4_INFO("alt: gp: %llu hp: %llu, rel: %f, glob: %f, home: %f", _global_pos_time, _home_time, msg.altitude_relative, global_alt, home.alt);
                }
 
                if (updated) {
```

LorenzMeier, 2016-12-25:
I can't repro but the SITL system is reproducing:
```
(vtol_new_1.txt) took too long to get to position lat:  47.397731863, lon:   8.543269296, alt:  15.00, xy off: 5.000000, z off: 5.000000, timeout: 600, index: 0, pos_d: 6.505292, alt_d: 473.084000, VTOL state: VTOL MC
```
Note the alt_d (altitude difference value) which is completely out of whack. That's because the tester and the vehicle couldn't agree on what home level is.

LorenzMeier, 2016-12-25:
@jgoppert I had it now with LPE that it would initialise at the baro altitude for home. Is this maybe the issue here?

dagar, 2017-02-16:
Duplicate https://github.com/PX4/Firmware/issues/6274?

dagar, 2017-04-11:
Fixed?

jgoppert, 2017-04-28:
Fixed on master.

closed 2017-04-28 by jgoppert