PX4/PX4-Autopilot #6170: Can't build on 14.04

CarlOlsson, 2016-12-25:
When I try to build for fmu-v4 I get the error:
```In file included from /home/carl/Wingtra/Standard_PX4/Firmware/src/systemcmds/hardfault_log/hardfault_log.c:56:0:
/home/carl/Wingtra/Standard_PX4/Firmware/src/modules/systemlib/px4_macros.h:91:54: error: expected ',' before ')' token
 #  define CCASSERT(predicate) static_assert(predicate)
                                                      ^
/home/carl/Wingtra/Standard_PX4/Firmware/src/systemcmds/hardfault_log/hardfault_log.c:137:1: note: in expansion of macro 'CCASSERT'
 CCASSERT(TIME_FMT_LEN == HEADER_TIME_FMT_LEN);
 ^
compilation terminated due to -Wfatal-errors.
make[3]: *** [src/systemcmds/hardfault_log/CMakeFiles/systemcmds__hardfault_log.dir/hardfault_log.c.obj] Error 1
make[2]: *** [src/systemcmds/hardfault_log/CMakeFiles/systemcmds__hardfault_log.dir/all] Error 2
```

CarlOlsson, 2016-12-25:
I just re-cloned the repo and now I get the error:
```
rsync: stat "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/apps/include/graphics/.tiff.h.nGf1yb" failed: No such file or directory (2)
rsync: rename "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/apps/include/graphics/.tiff.h.nGf1yb" -> "apps/include/graphics/tiff.h": No such file or directory (2)
rsync: stat "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/apps/nshlib/.Kconfig.giwYAn" failed: No such file or directory (2)
rsync: rename "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/apps/nshlib/.Kconfig.giwYAn" -> "apps/nshlib/Kconfig": No such file or directory (2)
rsync: stat "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/configs/olimex-strp711/src/.str71_spi.c.bsXFpE" failed: No such file or directory (2)
rsync: rename "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/configs/olimex-strp711/src/.str71_spi.c.bsXFpE" -> "nuttx/configs/olimex-strp711/src/str71_spi.c": No such file or directory (2)
rsync: stat "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/configs/stm32f429i-disco/nsh/.setenv.sh.MzDQbJ" failed: No such file or directory (2)
rsync: rename "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/configs/stm32f429i-disco/nsh/.setenv.sh.MzDQbJ" -> "nuttx/configs/stm32f429i-disco/nsh/setenv.sh": No such file or directory (2)
rsync: stat "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/drivers/sensors/.zerocross.c.vR1c9R" failed: No such file or directory (2)
rsync: rename "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/drivers/sensors/.zerocross.c.vR1c9R" -> "nuttx/drivers/sensors/zerocross.c": No such file or directory (2)
rsync: failed to set permissions on "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/net/icmpv6/.icmpv6_input.c.8iXJ0W": No such file or directory (2)
rsync: rename "/home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/build_px4fmu-v4_default/px4fmu-v4/NuttX/nuttx/net/icmpv6/.icmpv6_input.c.8iXJ0W" -> "nuttx/net/icmpv6/icmpv6_input.c": No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1183) [sender=3.1.0]
make[3]: *** [nuttx_copy_px4fmu-v4.stamp] Error 23
make[2]: *** [CMakeFiles/nuttx_patch_Fixed-Shadow-wanings.patch-px4fmu-v4.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  3%] Applying /home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/nuttx-patches/ctype_cctype_.patch
[  3%] Built target nuttx_patch_ctype_cctype_.patch-px4fmu-v4
[  3%] Applying /home/carl/Wingtra/Standard_PX4_newnuttx/Firmware/nuttx-patches/serial_dma_hotfix.patch
[  3%] Built target nuttx_patch_serial_dma_hotfix.patch-px4fmu-v4
[  3%] Built target xml_gen
make[1]: *** [all] Error 2
make: *** [px4fmu-v4_default] Error 2
```

kd0aij, 2016-12-25:
this looks like a duplicate of https://github.com/PX4/Firmware/issues/6128

CarlOlsson, 2016-12-25:
Yea I wasn't sure, please feel free to close the issue if it is

kd0aij, 2016-12-25:
if a ninja build works for you, then it's the same issue

LorenzMeier, 2016-12-26:
@dagar Yet another issue.

dagar, 2016-12-26:
Something weird with the nuttx copy (rsync). Trying to reproduce.

dagar, 2016-12-27:
@CarlOlsson should be resolved on master. Can you verify?

CarlOlsson, 2016-12-27:
Yes it works now, thanks!

closed 2016-12-27 by CarlOlsson