PX4/PX4-Autopilot #6175: LPE fusion issue in latest master

ecmnet, 2016-12-26:
@jgoppert With the latest (https://github.com/PX4/Firmware/commit/66b9ee2d24c0c91900fb270f4996305d8bac9b13) changes of LPE (fusion mask, etc), I have some serious issues. Z estimation does not take LIDAR measurements into account. LIDAR sometimes is reported as ok, but mostly faults with no (at least reported) recovery. Raw lidar measurements however are ok without any dropouts. 
I have set the fusion mask to 148 in order to disable AGL, GPS, vision Yaw and Flow as I purely rely on vision position estimates. LPE_VIS_Z and LPE_BARO I have set to maximum (to rely on LIDAR only). 
As a result, I get a Z estimation that seems to rely on the accelerator measurements only.

jgoppert, 2016-12-26:
@ecmnet The lidar still fuses when faulted now, so very odd. Can you upload a log?

ecmnet, 2016-12-27:
@jgoppert Currently I have no log as I reverted to stable. But I will provide a log asap. The issue occurs, as far as I remember also, if I use the standard bit mask (setting baro to 100). The resulting graph looked as if Z is a velocity and not a position.

ecmnet, 2016-12-28:
@jgoppert What I did: I connected my vehicle via USB and lifted it about 20-40cm above ground manually. LPE fusion mask is 247 in this case.

The log: http://logs.px4.io/plot_app?log=85d9ddb0-8046-4c80-b6e8-a13e3ff74e51

Additionally, the same test with 1.5.2:

The log: http://logs.px4.io/plot_app?log=682c67ac-1457-4b81-b2e5-6f2358727765

jgoppert, 2017-04-28:
Resolved on master.

closed 2017-04-28 by jgoppert