PX4/PX4-Autopilot #6190: Need to structure parallel builds on Semaphore better

LorenzMeier, 2016-12-27:
@dagar The build errors we keep getting with Semaphore seem to be a race condition inherent to Docker:
https://github.com/docker/docker/issues/21215

Either the 2016-12-26 image is bad (its the one it was trying to fetch) or we run into a race condition here.

dagar, 2016-12-27:
It looks like semaphore parallel jobs must be on the same machine. I'll try centralizing the pull in the setup section.

dagar, 2016-12-27:
Nope, it's the actually docker run. This really limits semaphore's usefulness for us.

dagar, 2016-12-27:
I've recombined firmware/bootloaders and tests/astyle.

dagar, 2016-12-27:
I think I'm at an impasse for getting this working on semaphore. I haven't found a way to work around parallel threads with any docker layer overlap, and all of our images now share base layers. This also means no splitting nuttx builds across multiple threads. 

I'll try contacting semaphore support, but I'm not expecting much.

dagar, 2017-01-14:
Nuttx builds currently split into qgc_firmware and alt_firmware and they're 12-15 minutes each.
https://github.com/PX4/Firmware/blob/master/Makefile#L204

We could try splitting them further.

closed 2017-01-14 by dagar