PX4/PX4-Autopilot #6193: GPS altitude spike feeding into altitude control loop despite GPS fusion disabled

LorenzMeier, 2016-12-27:
@priseborough @tumbili This is an interesting log. It shows the altitude setpoint spiking at the same time as GPS although the GPS accuracy is completely beyond any fusion threshold.

Is the GPS altitude fused outside of the main EKF fusion?
http://review.px4.io/plot_app?log=e831da16-47ba-4cec-9ec0-aa5fe039bba9

priseborough, 2016-12-27:
There is supposed to be no fusion of GPS in the ecl library other than in the main EKF. To verify this I've run the log through the ekf ulog analyser and they show that the EKF starts using GPS towards the end of flight. Here are the vertical innovations:
[image]

You can see the noise and spikes in baro innovation early in flight which correlate with the aggressive manoeuvres. Other than that the EKF tracks the baro well and the local position z agrees with the EKF state:

[image]

There is nothing in the EKF vertical position output that would explain the change in local position set-point. The local position vz, the EKF vz state and the derivative of the local position z are consistent and track the demanded vertical velocity:

[image]

The correlation between the large jump in logged GPS height and vehicle local position set-point log is clear and it it is consistent with an un-commanded jump in velocity set-point.
[image]

Checking the reference altitude in the local position message shows that until the EKF starts using GPS, the vehicle_local_position.ref_alt message is being set to the current GPS alt. This is because the ekf2 module is using the EKF origin to set vehicle_local_position.ref_alt and until the EKF has started GPS navigation, this moves with the flight vehicle.

I believe this to be the source of the problem as the position controller uses the altitude reference when calculating the set-point. 

This can be solved by:

1) Setting the local position ref_alt to the pressure altitude of the EKF origin until the EKF passes GPS checks and the global position estimate is valid. The problem with this method is that if the EKF starts using GPS mid-flight, then there will be a jump in vehicle_local_position.ref_alt at that time. 

2) Prevent jumps in set-point when gaining GPS by modifying the position controller to not use the ref_alt when operating in a non-auto control mode.

priseborough, 2016-12-27:
https://github.com/PX4/Firmware/pull/6197

LorenzMeier, 2017-01-02:
Closing since the PR tracks it.

PR #6459 (2017-05-01): Prevent height set-point variations due to bad GPS
Fixes: #6193

TODO

Code review
Compatibility testing with LPE
closed 2017-01-02 by LorenzMeier