PX4/PX4-Autopilot #6198: Stream Rate for Specific Mavlink messages

edsellar, 2016-12-28:
I've read through the other posting about similar streaming issues but none answered this.

How do I set the stream rate of specific messages to on Serial 2 to stream at a specific hz rate because all I can do is limit them to a specific data rate with -r (mavlink stream -d /dev/ttyS2 -s ALTITUDE -r X)

Reason being that some of the newer radio modems I've been testing are geared towards the software side of altering APM parameters (stream rates mainly) but while these are just mavlink messages coming down to the software ground station that we need to control the rates, the parameters don't all translate over. I can't make it connect but not work, When I do connect successfully, params dont download on QGC and I get non-sensical readings (RSSI at 253% [probably a raw # instead of %, voltage of 40v [only using usb], etc). 

I've tried on stable and master. The need for Telem2 to drop down to 19200 baud was crucial in this but just the start of how to make the system more compatible.

The quick list which is pretty standard across quite a few different vendors of radio modems that only have APM settings and not have PX4 specific ones to set are (I wrote them out below which what they entail):

BRD_SER1_RTSCTS 0
SERIAL1_BAUD 19
SERIAL1_PROTOCOL 1
SR1_EXT_STAT 1
SR1_EXTRA1 5
SR1_EXTRA2 5
SR1_EXTRA3 1
SR1_PARAMS 8
SR1_POSITION 3
SR1_RAW_CTRL 1
SR1_RAW_SENS 1
SR1_RC_CHAN 0

This specific testing is using bi-directional telemetry with a new 1.5w dragonlink RC/Telem system. 
Back to the parameters needed to be altered:

BRD_SER1_RTSCTS 0    - **Can do when adding the link in settings of QGC**
SERIAL1_BAUD 19  - **Can do with master and setting baud to 19200 of Telem2**
SERIAL1_PROTOCOL 1 - **Can set to only go through mav ver 1 through Params**

Then it gets tricky:

_SR1_EXT_STAT 1_: 1 = 1hz and contains the following:
     SYS_STATUS - APM
		SYS_STATUS - PX4 - Match
	MEMINFO - APM
              Couldn't match for PX4
	MISSION_CURRENT - APM
		MISSION_ITEM - ? Maybe match for PX4
	GPS_RAW_INT - APM
		GPS_RAW_INT - PX4 Match
	NAV_CONTROLLER_OUTPUT - APM
		NAV_CONTROLLER_OUTPUT - PX4 - Match
	LIMITS_STATUS - APM
                 Couldn't match for PX4

_SR1_EXTRA1 5_ (set to 5hz):
      ATTITUDE - APM and PX4 match

_SR1_EXTRA2 5_ (set to 5hz):
      VFR_HUD - APM and PX4 match

_SR1_EXTRA3 1_ (set to 1hz):
        AHRS - APM
                 Couldn't find a match.
        HWSTATUS - APM
              Couldn't find a match.
        SYSTEM_TIME - Match on PX4

_SR1_PARAMS 8_ (8hz)
       - Params stream rate on APM - Couldn't find equiv on PX4

_SR1_POSITION 3_
     - GLOBAL_POSITION_INT - MATCH

_SR1_RAW_CTRL 1_
   - RC_CHANNELS_SCALED - APM - No match for PX4

_SR1_RAW_SENS 1_ (all apm - couldn't match them)
   - RAW_IMU
   - SCALED_IMU2
   - SCALED_PRESSURE
   - SENSOR_OFFSETS 

_SR1_RC_CHAN 0_
     SERVO_OUTPUT_RAW - APM
         - SERVO_OUTPUT_RAW_0  on PX4?
         - SERVO_OUTPUT_RAW_1  on PX4?
     RC_CHANNELS - APM
         - Couldn't match

Ok so that's quite a bit but when the hardware developer doesn't take any steps to try and accommodate more than just one group of users even though it should be easy for them to do and completely possible, it's on us to figure out which is totally fine but a bit tricky. I set as many of the above that match via NSH but still only get their rates altered with -r. See the screenshot of what it looks like when I connect (and I alter them via NSH and then connect through a bluetooth telem connection because I know when connected through USB the data rate is ignored). This is good because would allow bi-directional telem over the RC channel at full bandwidth and then it is one less thing (radio modem), to fit in our usually small machines but overall it seems to be a reoccurring set of parameters to alter from several different modems/quasi modems for control.

I want to keep trying but some/any help would be really great!

[image]

edsellar, 2016-12-28:
Going to revert back on the firmware side of the modem im testing and retest all this and get back with results but still need to find what all the Mav messages correlate to.

LorenzMeier, 2016-12-28:
I think the bit that you're missing is that the stream rate limiting is indeed what you want - PX4 only updates information as it changes, never when there is no change. So if you set the rate limit you're still going to get every update, but if you set the GPS rate to 10 Hz but the GPS receiver only updates at 5 Hz you will get just 5 - because that is reflecting reality.

edsellar, 2016-12-28:
That is sort of what I was trying to understand. The -r parameter limits the data-rate but how do I correlate that from the data-rate perspective I can control, to the hz perspective that this is expected? It is apples to oranges from the settings perspective but apples to apples in that it is still trying to achieve the same thing, not flooding the data-link with more than it can handle by slowing the reporting down/amount of data it will pass through. I imagine by monitoring mavlink status and the total available link speed and making sure that it is not surpassed in aggregate is the path? Surely there is a way I can calculate this instead of trial and error. Settings and reality are indeed irrelevant if reality is slower as it won't hit the capped rate but watching the mav stream data-rate and making sure it does not kick out too many lost packet to go over the limit seems as though it will work but definitely far from ideal. Another way? Thanks.