PX4/PX4-Autopilot #6200: Switch Mission tests CI to Dronekit

LorenzMeier, 2016-12-28:
Continuing this when upstream repos are updated:
https://github.com/PX4/Firmware/pull/6199#pullrequestreview-14548636

The goal is to make CI more reproducible on the desktop and to reduce the dependencies.

AndreasAntener, 2016-12-28:
I'm for moving away from any "third party" lib actually. And only include one or two basic compatibility tests for dronekit and mavros.

dagar, 2017-01-02:
What's the overall plan here? Move all mission testing to Dronekit and drop ROS?
Let me know if I can help getting the dependencies in docker or getting the builds running.

Once we have something basic running I'll add code coverage so it's obvious what tests should be added to reach as much code as possible.
 
 - FW mission takeoff, waypoints, loiter, land
 - mission with geofence
 - etc

I'd also like to make sure it's very easy to run these locally with or without a container, eg `make mission_tests` and the ability to specify a particular mission.

dagar, 2017-01-12:
@sanderux px4io/px4-dev-simulation is updated on docker hub with the latest pymavlink and dronekit. https://hub.docker.com/r/px4io/px4-dev-simulation/

What are the next steps for mission tests? Can we come up with an easy way to run them from the makefile like `make mission_tests`? We could use ctest or some other wrapper to run each mission individually and report pass/fail.
Then I'll add a job on jenkins that simply runs `make mission_tests` from within px4io/px4-dev-simulation.

sanderux, 2017-01-12:
the mission test script starts when gazebo_tandard_vtol is running
```
cd integrationtests/python_src/px4_it/dronekit
python MissionCheck.py
```

I guess a makefile target would work too

dagar, 2017-01-12:
Ok, so we just need to a way to organize starting multicopter, plane, vtol and running the appropriate mission file.

sanderux, 2017-01-12:
It can take a mission file from the command line:
```
$ python MissionCheck.py --help
usage: MissionCheck.py [-h] [-c CONNECT] [-f FILENAME] [-t TIMEOUT]
                       [-a ALTRAD]

optional arguments:
  -h, --help            show this help message and exit
  -c CONNECT, --connect CONNECT
                        connection string
  -f FILENAME, --filename FILENAME
                        mission filename
  -t TIMEOUT, --timeout TIMEOUT
                        execution timeout
  -a ALTRAD, --altrad ALTRAD
                        altitude acceptance radius
```

AndreasAntener, 2017-01-13:
@dagar @sanderux Makefile/cmake integration would easy and preferable so the entry point is the same as for normal development, have a look how the unittests on posix are run (via ctest target execution `sitl_run.sh`). So it just needs a sitl_run script that also fires off the test (which can be a plain python `unittest`) after booting SITL

AndreasAntener, 2017-01-13:
Oh I see Daniel already is chief here ;): https://github.com/PX4/Firmware/blob/48e7c784e793957cb50ce62aa890438273d90048/src/firmware/posix/sitl_tests.cmake

My main concern would be test naming so it's clear which test name a user has to pass on via the make target

dagar, 2017-01-13:
I was just debating adding them to the existing tests or having something separate.
So then we'd be able to easily run them separately.
 - unit tests that run within a PX4 environment
 - standalone unit tests (one sad param gtest)
 - mission tests

We should try and name them descriptively and make it easy to run both in and outside of a container.

Step 1 - script that starts headless sitl gazebo in the background, runs the mission test, and then shuts down cleanly. Then we wrap ctest around it and make it as easy as possible to add more, including different airframes.

AndreasAntener, 2017-01-13:
Right.
For the mission tests we could iterate over a test data folder that contains mission files and generate add_test commands. Possibly with a naming convention to specify which sitl to run.

closed 2018-01-17 by dagar