PX4/PX4-Autopilot #6207: sensor cal problem with SITL?

kd0aij, 2016-12-28:
Just installed gazebo 7 on Ubuntu 15.10:
seems to start up OK, but won't arm.

"make posix_sitl_default gazebo" 
PREFLIGHT FAIL: ACCELS INCONSISTENT - CHECK CAL

https://gist.github.com/kd0aij/5668c0e2824d416f49626424e24f3343

LorenzMeier, 2016-12-29:
Yes, this was indeed wrong. I have a fix for you here: https://github.com/PX4/Firmware/pull/6213

Can you test?

kd0aij, 2016-12-30:
Still get the "accels inconsistent" message with both jmavsim and gazebo7 (tried both posix_sitl_default and posix_sitl_ekf2). Quadcopter flies as expected in jmavsim, but rotors don't spin in gazebo7 even though I'm now able to arm. 

Is this "user error" or does gazebo not like Ubuntu 16.04?

dagar, 2017-04-11:
I believe this is fixed, I've been using gazebo sitl.

closed 2017-04-11 by dagar