PX4/PX4-Autopilot #6210: Spektrum RC radio control - channels mixed up?

sunshine-yan, 2016-12-29:
Hi devs,
I have updated the latest qcom_flight_controller_hexagon_sdk_add_on.zip,BSP.but when I move the throttle/yaw stick . Unfortunately, in the QGroundControl channel monitor:
At present, only 1,2,3-channel data can be received.
Apart from the right stick (roll/pitch),left stick and a second gear switch,the others  isn't showing up in any channels at all.
Has anyone seen this issue before and if so, is there a suggested fix to get things back to normal?
Thanks for your help!
Ed!

Stifael, 2016-12-29:
I think I did  not entirely understand your issue:
What happens when you move the throttle/yaw stick?
I have a spektrum transmitter and receiver. for me it is as followed:
channel 3: throttle
channel 4: yaw
channel 2: pitch
channel 1: roll
channel 5: flight mode

LorenzMeier, 2016-12-29:
I think he is not using the open source Spektrum driver you're using @Stifael. We need to finish our work on Snapdragon and document it - that will make it a lot easier for people.

Stifael, 2016-12-29:
I see. I was just wondering, because he shouldnt be able to use it out of the box with the current firmware.

sunshine-yan, 2016-12-30:
@Stifael @LorenzMeier How can i use the open source Spektrum driver? At present, only 1,2,3 channel data can be received  when i move the throttle/yaw stick.

LorenzMeier, 2016-12-30:
@Stifael Can you please send a pull request with the update to the startup script which starts it?

Stifael, 2016-12-30:
@sunshine-yan:
Currently I dont have my laptop, but you can do it yourself: just go to 
posix-configs/eagle/flight/px4.config
and add 
spektrum_rc start

Stifael, 2016-12-30:
I can make a PR hopefully tmr or the latest this sunday. Or u can do it youself (the driver is already added in the master)

Stifael, 2016-12-30:
And the doko will be updated as soon as well

sunshine-yan, 2017-01-06:
@Stifael 
Thanks for your help! This problem is sovled by following your suggestion.

closed 2017-01-06 by Stifael