PX4/PX4-Autopilot #6216: Wind estimation not working?

vvranjek, 2016-12-29:
WIND_COV sends only zeros except for altitude. I am on v1.5.2. Is it supposed to work or is this feature not supported yet?

CarlOlsson, 2016-12-29:
It should work, could you upload a log file?

vvranjek, 2016-12-29:
Sure, here it is. 

[12_36_37.px4log.zip](https://github.com/PX4/Firmware/files/677584/12_36_37.px4log.zip)

CarlOlsson, 2016-12-30:
The reason to why you don't have any estimated wind states is because you have the airspeed threshold equals to zero. (EKF2_ARSP_THR = 0) If you set this one to something higher, e.g. 7 m/s you will start estimating the wind. 
However, before you do that you need to solve some estimator issues. Your position and velocity states are diverging:
[image]
A test ratio above 1 indicates that the estimator stops fusing the sensor and you can see several resets when the test ratio abruptly jumps back to 0.

You also have some throttle to magnetometer interference, if your power cables already are as far as possible from the mag you can try twisting them.

One last thing, for uploading logs you can use [this website](http://logs.uaventure.com/) with the old logger (SYS_LOGGER = 0) or [this website](http://logs.px4.io/) for the new logger (SYS_LOGGER = 1)

vvranjek, 2016-12-30:
Wow, this is super useful info, I can't thank you enough for it. I know I had some estimator issues and I solved it by recalibrating the sensors, but I am not sure which flights were made with bad calibrations. But I had no idea about the magnetometer, makes sense since I do have power cables very close to sensors/electronics. Thanks for pointing it out.

I guess I will start using ulog. Are there any GUI tools available for displaying the ulog data. I found the pyulog but I don't think it is capable of drawing graphs like flightPlot?

I just found a lot of new documentation on debugging that I believe wasn't there months ago, which is great! Do you have any suggestions on best ways for debugging and analyzing flights for fine tuning the parameters and overall debugging? How/where were the graphs above made?

And again, thanks a lot for all your help!

kd0aij, 2016-12-30:
@vvranjek Documentation for the new web-based "Flight Review" is  here: http://dev.px4.io/flight_log_analysis.html#flight-review

pyulog can convert .ulg files to csv files (ulog2csv.py) for subsequent analysis/display in Octave or Matlab

vvranjek, 2016-12-30:
Ok, thanks! I am especially interested in how @CarlOlsson created the 'Test ratio' graphs? Otherwise, this issue is solved and can be closed.

CarlOlsson, 2017-01-02:
No problem, glad to help.
The plots above are from Matlab. For general debugging I would try MAVExplorer or Flightplot for .px4log or flight review (you can also run it locally) for .ulg. If you miss functionality in flight review everyone is thankful if you point it out. 

For the wind estimation you can also set the EKF2_FUSE_BETA parameter to 1 in order to also estimate the wind from synthetic sideslip measurements.

closed 2017-01-02 by kd0aij

bellahduan, 2017-10-10:
@CarlOlsson hi, Sorry to disturb you . but i am confused . the log download from QGC is ***.px4log . but it can not use to analysis . Is there any way to solve the problem?
thank u so much .