PX4/PX4-Autopilot #6222: Save `.ulg' failed with POSIX and MAVROS?

weiweikong, 2016-12-31:
I tried to use current master version to test the my algorithms by `.ulg` log file using
```
 roslaunch px4 mavros_posix_sitl.launch
```
The system worked well as follwoing
```
... logging to /home/amax/.ros/log/04bea1e2-cf0b-11e6-a4ea-002590fdcfd7/roslaunch-amax-X10DRi-10770.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://amax-X10DRi:49418/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['safety_area', '...
 * /mavros/plugin_whitelist: []
[... 177 lines cut ...]
INFO  [commander] Takeoff detected
[ INFO] [1483155751.347595790, 57.438000000]: FCU: Takeoff detected
INFO  [tone_alarm] positive
INFO  [tone_alarm] neutral
[ WARN] [1483155763.333991619, 69.406000000]: CMD: Unexpected command 400, result 0
INFO  [logger] closed logfile, bytes written: 4357056
[ INFO] [1483155763.348739377, 69.420000000]: FCU: DISARMED by arm/disarm component command
INFO  [local_position_estimator] [lpe] land init
[ INFO] [1483155763.898135845, 69.968000000]: FCU: [lpe] land init
INFO  [commander] Landing detected
[ INFO] [1483155765.399206218, 71.468000000]: FCU: Landing detected

```
I could ARM and DISARM correctly through MAVROS, but the system did not save the log file correctly even though the terminal told me
```
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2016-12-31/03_41_36.ulg
```
I checked that the folder `Firmware/build_posix_sitl_default/tmp/rootfs/fs/microsd/log/2016-12-31` was clean. Also, I searched the whole `Fimware` folder by
```
✔ amax@amax-X10DRi|~/Workspace/53.PX4_New/Firmware [master|✚ ] 
11:43:21 $ find . -name "*ulg*"
✔ amax@amax-X10DRi|~/Workspace/53.PX4_New/Firmware [master|✚ ] 
11:43:24 $ 
```
but got nothing.

bkueng, 2017-01-03:
I don't have ROS installed, so I cannot test it. It seems ROS uses [$ROS_HOME](http://wiki.ros.org/ROS/EnvironmentVariables#ROS_HOME) as working directory, so you can start to look in there if you find an .ulg file.

weiweikong, 2017-01-04:
@bkueng Thank you for your reply. 

In my opinion, the `$ROS_HOME` folder only recorded the related terminal feedback. I tried and checked the folder, but there are 
```
✔ amax@amax-X10DRi|~/.ros/log/04bea1e2-cf0b-11e6-a4ea-002590fdcfd7 
07:55:34 $ ls
keyboard-1-stdout.log
master.log
roslaunch-amax-X10DRi-10770.log
roslaunch-amax-X10DRi-11634.log
rosout-1-stdout.log
rosout.log
vehicle_spawn_amax_X10DRi_10770_1154193950404211463-5.log
```
In previous tests, without MAVROS, such as running `make posix_sitl_default gazebo` instead of ` roslaunch px4 mavros_posix_sitl.launch`,  the `.px4log` and `.ulg` saved correctly in the ` build_posix_sitl_default/rootfs/fs/microsd/log/` folder whith is independent of `$ROS_HOME` folder.

mhkabir, 2017-04-03:
Please reopen if this is still an issue. It works fine for me locally.

closed 2017-04-03 by mhkabir

NRottmann, 2017-06-12:
@weiweikong or @mhkabir 
I am facing the same Issue. I am running my simulation with mavros and .ulg is not saved correctly. If I instead make posix_sitl_default gazebo then it stores the log files correctly. How did you solve the problem?
Best Nils