PX4/PX4-Autopilot #6223: USB mass storage support

western-co-at, 2016-12-31:
Hello PX4 team
It would be nice if the micro sd on the PX4 FMU can be accessed by USB to modify startup scripts or download log files.

I tried already to add the usbmsc driver (current master with NuttX 7.18):
https://github.com/western-co-at/Firmware/commit/a832c0a0c4c51e1e1f4231a5aa969d11ab7a5f4c

The mass storage driver can be activated by the following commands:
`nsh> mavlink stop-all`
`WARN  [mavlink] waiting for instances to stop`
`..WARN  [mavlink] exiting channel 0`
`....WARN  [mavlink] exiting channel 1`
`.....................WARN  [mavlink] exiting channel 2`
`WARN  [mavlink] all instances stopped`
`nsh>`
`nsh>`
`nsh> serdis`
`serdis: Disconnected`
`nsh> umount /fs/microsd`
`nsh> msconn`
`mcsonn_main: Creating block drivers`
`mcsonn_main: Configuring with NLUNS=1`
`mcsonn_main: handle=2000d6a0`
`mcsonn_main: Bind LUN=0 to /dev/mmcsd0`
`mcsonn_main: Connected`
`nsh>`

In case the FMU boots a multicopter with EKF2 enabled the USB mass storage is working 70% of time:
`[52183.080054] usb 5-2: new full-speed USB device number 125 using uhci_hcd`
`[52183.257156] usb 5-2: New USB device found, idVendor=584e, idProduct=5342`
`[52183.257162] usb 5-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3`
`[52183.257165] usb 5-2: Product: USBdev Storage`
`[52183.257168] usb 5-2: Manufacturer: NuttX`
`[52183.257171] usb 5-2: SerialNumber: 0101`
`[52183.265246] usb-storage 5-2:1.0: USB Mass Storage device detected`
`[52183.273625] scsi host38: usb-storage 5-2:1.0`
`[52184.276212] scsi 38:0:0:0: Direct-Access     NuttX    USBdev Storage   0101 PQ: 0 ANSI: 2`
`[52184.276846] sd 38:0:0:0: Attached scsi generic sg2 type 0`
`[52184.283209] sd 38:0:0:0: [sdb] 15802368 512-byte logical blocks: (8.09 GB/7.54 GiB)`
`[52184.487256] sd 38:0:0:0: [sdb] Write Protect is off`
`[52184.487268] sd 38:0:0:0: [sdb] Mode Sense: 0f 00 00 00`
`[52184.729253] sd 38:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA`
`[52185.152315]  sdb: sdb1`
`[52185.575241] sd 38:0:0:0: [sdb] Attached SCSI removable disk`
`[52190.039430] FAT-fs (sdb1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.`

If the rcS boot script is aborted with a blank /fs/microsd/etc/rc.txt script the USB mass storage isn't working anymore:
`[49093.000089] usb 5-2: new full-speed USB device number 119 using uhci_hcd`
`[49093.174388] usb 5-2: New USB device found, idVendor=584e, idProduct=5342`
`[49093.174394] usb 5-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3`
`[49093.174397] usb 5-2: Product: USBdev Storage`
`[49093.174399] usb 5-2: Manufacturer: NuttX`
`[49093.174402] usb 5-2: SerialNumber: 0101`
`[49093.183478] usb-storage 5-2:1.0: USB Mass Storage device detected`
`[49093.188121] scsi host36: usb-storage 5-2:1.0`
`[49094.192390] scsi 36:0:0:0: Direct-Access     NuttX    USBdev Storage   0101 PQ: 0 ANSI: 2`
`[49094.193630] sd 36:0:0:0: Attached scsi generic sg2 type 0`
`[49094.199882] sd 36:0:0:0: [sdb] 15802368 512-byte logical blocks: (8.09 GB/7.54 GiB)`
`[49094.402269] sd 36:0:0:0: [sdb] Write Protect is off`
`[49094.402277] sd 36:0:0:0: [sdb] Mode Sense: 0f 00 00 00`
`[49094.604239] sd 36:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA`
`[49095.225191] sd 36:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE`
`[49095.225202] sd 36:0:0:0: [sdb] tag#0 Sense Key : Medium Error [current] `
`[49095.225209] sd 36:0:0:0: [sdb] tag#0 Add. Sense: Unrecovered read error`
`[49095.225216] sd 36:0:0:0: [sdb] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00`
`[49095.225222] blk_update_request: critical medium error, dev sdb, sector 0`
`[49095.225228] Buffer I/O error on dev sdb, logical block 0, async page read`
`[49095.630118] sd 36:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE`
`[49095.630126] sd 36:0:0:0: [sdb] tag#0 Sense Key : Medium Error [current] `
`[49095.630131] sd 36:0:0:0: [sdb] tag#0 Add. Sense: Unrecovered read error`
`[49095.630136] sd 36:0:0:0: [sdb] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00`
[... 40 lines cut ...]
`[49098.466810] sd 36:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE`
`[49098.466817] sd 36:0:0:0: [sdb] tag#0 Sense Key : Medium Error [current] `
`[49098.466822] sd 36:0:0:0: [sdb] tag#0 Add. Sense: Unrecovered read error`
`[49098.466826] sd 36:0:0:0: [sdb] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00`
`[49098.466830] blk_update_request: critical medium error, dev sdb, sector 0`
`[49098.466834] Buffer I/O error on dev sdb, logical block 0, async page read`
`[49098.872770] sd 36:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE`
`[49098.872779] sd 36:0:0:0: [sdb] tag#0 Sense Key : Medium Error [current] `
`[49098.872783] sd 36:0:0:0: [sdb] tag#0 Add. Sense: Unrecovered read error`
`[49098.872788] sd 36:0:0:0: [sdb] tag#0 CDB: Read(10) 28 00 00 00 00 18 00 00 08 00`
`[49098.872792] blk_update_request: critical medium error, dev sdb, sector 24`
`[49099.684692]  sdb: unable to read partition table`
`[49100.106691] sd 36:0:0:0: [sdb] Attached SCSI removable disk`
`[49100.918541] scsi_io_completion: 3 callbacks suppressed`
`[49100.918556] sd 36:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE`
`[49100.918564] sd 36:0:0:0: [sdb] tag#0 Sense Key : Medium Error [current] `
`[49100.918570] sd 36:0:0:0: [sdb] tag#0 Add. Sense: Unrecovered read error`
`[49100.918578] sd 36:0:0:0: [sdb] tag#0 CDB: Read(10) 28 00 00 f1 1f 80 00 00 08 00`
`[49100.918581] blk_update_request: 3 callbacks suppressed`
`[49100.918586] blk_update_request: critical medium error, dev sdb, sector 15802240`
`[49100.918591] buffer_io_error: 3 callbacks suppressed`
`[49100.918595] Buffer I/O error on dev sdb, logical block 1975280, async page read`

At the nsh console there is no error output in case it is not working.
I tried it already with sandisk and swissbit µSD cards.

LorenzMeier, 2017-01-02:
Would indeed be interesting.

I would propose to modify the ROMFS/px4fmu_common/init.d/rcS script to a minimal case (just start the storage device) and validating that. Check the RAM usage before and after (run `free`).

Are you aware that you can download log files via QGroundControl already over USB serial?

western-co-at, 2017-01-03:
I tried to reduce the script step-by-step but I could figure out which commands/apps are important because it's not working reliably also after a full boot sequence.

I'm aware that I can download the logs via mavlink. But I had sometimes problems with the sd card and had to format it. So it would also be possible to format it via the px4fmu unsing the SDFormatter tool.

In case it is working I get following RAM usage (with debugs enabled):
`nsh> msconn`
`msconn`
`After usbtrace_enable():`
`              total       used       free    largest`
`Before:     236704     139232      97472      59104`
`After:      236704     139232      97472      59104`
`mcsonn_main: Creating block drivers`
`After boardctl(BOARDIOC_USBDEV_CONTROL):`
`              total       used       free    largest`
`Before:     236704     139232      97472      59104`
`After:      236704     139232      97472      59104`
`mcsonn_main: Configuring with NLUNS=1`
`mcsonn_main: handle=2000ecd0`
`After usbmsc_configure():`
`              total       used       free    largest`
`Before:     236704     139232      97472      59104`
`After:      236704     139504      97200      59104`
`Change:        272 allocated`
`mcsonn_main: Bind LUN=0 to /dev/mmcsd0`
`After usbmsc_bindlun():`
`              total       used       free    largest`
`Before:     236704     139504      97200      59104`
`After:      236704     140032      96672      59104`
`Change:        528 allocated`
`After usbmsc_exportluns():`
`              total       used       free    largest`
`Before:     236704     140032      96672      59104`
`After:      236704     145360      91344      59104`
`Change:       5328 allocated`
`mcsonn_main: Connected`
`After MS connection:`
`              total       used       free    largest`
`Before:     236704     145360      91344      59104`
`After:      236704     145360      91344      59104`

With the aborted rcS script and not working usbmsc it returns:
`nsh> msconn`
`msconn`
`After usbtrace_enable():`
`              total       used       free    largest`
`Before:     236704      23472     213232     168976`
`After:      236704      23472     213232     168976`
`mcsonn_main: Creating block drivers`
`After boardctl(BOARDIOC_USBDEV_CONTROL):`
`              total       used       free    largest`
`Before:     236704      23472     213232     168976`
`After:      236704      23472     213232     168976`
`mcsonn_main: Configuring with NLUNS=1`
`mcsonn_main: handle=10004a10`
`After usbmsc_configure():`
`              total       used       free    largest`
`Before:     236704      23472     213232     168976`
`After:      236704      23744     212960     168976`
`Change:        272 allocated`
`mcsonn_main: Bind LUN=0 to /dev/mmcsd0`
`After usbmsc_bindlun():`
`              total       used       free    largest`
`Before:     236704      23744     212960     168976`
`After:      236704      24272     212432     168976`
`Change:        528 allocated`
`After usbmsc_exportluns():`
`              total       used       free    largest`
`Before:     236704      24272     212432     168976`
`After:      236704      29264     207440     168976`
`Change:       4992 allocated`
`mcsonn_main: Connected`
`After MS connection:`
`              total       used       free    largest`
`Before:     236704      29264     207440     168976`
`After:      236704      29264     207440     168976`

For a short test I have also increased the CONFIG_USBMSC_SCSI_STACKSIZE - but without success.

If following defs are set (replacing the defaults like it is done e.g. for the stm3210e-eval or hymini-stm32v boards):
`CONFIG_USBMSC_EPBULKOUT=2`
`CONFIG_USBMSC_EPBULKIN=5`
the msconn command return with following error code:
`nsh> msconn`
`After usbtrace_enable():`
`              total       used       free    largest`
`Before:     236704     143200      93504      58800`
`After:      236704     143200      93504      58800`
`mcsonn_main: Creating block drivers`
`After boardctl(BOARDIOC_USBDEV_CONTROL):`
`              total       used       free    largest`
`Before:     236704     143200      93504      58800`
`After:      236704     143200      93504      58800`
`mcsonn_main: Configuring with NLUNS=1`
`mcsonn_main: handle=1000c220`
`After usbmsc_configure():`
`              total       used       free    largest`
`Before:     236704     143200      93504      58800`
`After:      236704     143440      93264      58800`
`Change:        240 allocated`
`mcsonn_main: Bind LUN=0 to /dev/mmcsd0`
`After usbmsc_bindlun():`
`              total       used       free    largest`
`Before:     236704     143440      93264      58800`
`After:      236704     143968      92736      58800`
`Change:        528 allocated`
`mcsonn_main: usbmsc_exportluns failed: 19`
Can't the EP use any numbers?

mhkabir, 2017-01-03:
@western-co-at I've taken the liberty to gate your terminal output with "```" to get GitHub to render it nicely :) Use it instead of single "`" for multiline outputs.

davids5, 2017-01-03:
The filesystem uses of the SD card is mutually exclusive with mounting the block driver to provide the scsi access to the SD card. The card has to be unmounted from the fs to be given to the scsi access via the MSC. 

The notion of a give to MSC driver and take from MSC driver has to be added and triggered. I have done this before with the VBUS detection and a single use, 1 file application, but I fear it is much more complicated with the PX4 usecase

closed 2017-10-04 by LorenzMeier

western-co-at, 2017-12-15:
I have seen that this ticket got closed and tried to test the USB mass storage driver of nuttx 7.22+/current master again by adding following lines to the defconfig:
```CONFIG_USBMSC_VENDORID=0x584e
CONFIG_USBMSC_VENDORSTR="NuttX"
CONFIG_USBMSC_PRODUCTID=0x5342
CONFIG_USBMSC_PRODUCTSTR="USBdev Storage"
CONFIG_USBMSC_VERSIONNO=0x0399
CONFIG_USBMSC_REMOVABLE=y
CONFIG_USBMSC_SCSI_PRIO=128
CONFIG_USBMSC_SCSI_STACKSIZE=2048
CONFIG_SYSTEM_USBMSC=y
```
In the NSH I entered following lines:
```nsh> mavlink stop-all
nsh> dataman stop
nsh> logger stop
nsh> umount /fs/microsd
nsh> serdis
nsh> msconn
nsh> msconn
mcsonn_main: Creating block drivers
mcsonn_main: Configuring with NLUNS=1
mcsonn_main: handle=10006470
mcsonn_main: Bind LUN=0 to /dev/mmcsd0
mcsonn_main: Connected
```
The microsd card is still not allways mountable. After a new reboot it may be mountable. Do I have to set another CONFIG switch/parameter (e.g. a special endpoint number) to get the msc driver working reliably?

dagar, 2017-12-15:
@western-co-at I'm not sure, but you could try asking on the NuttX mailing list.

I'm interested in this functionality, but I don't know how to make it fit into PX4.

davids5, 2017-12-21:
1) There are not enough endpoints in HW on the F4 to do MSC and CDC/ACM. 2) my comment above still applies