PX4/PX4-Autopilot #6225: should sf1xx driver be removed?

lovettchris, 2016-12-31:
I noticed [this checkin ](https://github.com/PX4/Firmware/commit/fa834497bfd61d9d23d14aa501013f22f064cd93) which seem to add sf1xx support to sf0x driver.  Should the sf1xx driver be removed then ?  It is confusing having both... which one should I use now for the sf11/c sensor?  I'm more confused by the fact that the changes in the above commit are so few, there seem to be a lot more differences in the sf1xx driver code...

ecmnet, 2017-01-02:
@lovettchris Sf0x driver supports serial connection, while sf1xx supports I2C connected devices as far as I know.

jgoppert, 2017-01-02:
Yes, that is correct sf0x (serial), sf1xx (i2c). We might want to rename them at some point to avoid this confusion and do some refactoring to make sure they operate in the same manner.

lovettchris, 2017-01-03:
Ah, ok, the i2c/serial difference wasn't clear from the names.  Would be more clear if those could be merged.  I need the i2c version since the i2c splitter means I don't have to use up a valuable serial port.  This works nicely with the Lightware firmware version "***  SF11/C  Rev 1.21  ***".

closed 2017-04-12 by dagar

fe7565, 2018-03-18:
I have the SF11/C Rev 1.1 firmware.    How can I update it to the 1.21 version without having to send it to South Africa?   Thank you for your assistance.

weiminshen99, 2018-07-05:
Dear Everyone,

Could someone clarify the status of this issue? I am connecting a new sensor SF10/a to Serial/4 port, and set SF0X = 2, but the sensor is not working (all readings are errors). Please help! Thanks!